msm8953-mainline/linux #23: Xiaomi-daisy: Ilitek ili7807-fdhplus display not working

apolemus, 2021-11-14:
With yesterday built kernel from here the ili7807-fhdplus display on my Xiaomi Mi A2 Lite (daisy) doesn’t come up (except for some blue artefacts). I’m using lk2nd and am able to ssh into the device (I’ll add the dmesg log after boot). Some time ago I already tested out the patches of #21 with the same outcome. Without the patches, it is not booting until ssh.

Am I missing something or is this still WIP?

[dmesg_daisy_20211113.log](https://github.com/msm8953-mainline/linux/files/7533923/dmesg_daisy_20211113.log)

alikates, 2021-11-14:
Hello, I think the kernel is okay and the issues is with lk2nd. The ili7807-fhdplus panel was tested with a similar driver for the fhd variant before creating the fhdplus variant driver. I forgot to change it in lk2nd.
So I'm gonna open a PR to lk2nd, but if you want it soon I'm writing a quick fix to my personal fork here https://gitlab.com/alikates/lk2nd

apolemus, 2021-11-14:
Hello,
thanks a lot for your time! I tried your lk2nd ‘daisy’ tree with last commit 2019f490 but the issue persists. I also tried to change `qcom,mdss_dsi_ili7807_fhd_video` to `qcom,mdss_dsi_ili7807_fhdplus_video` (as lk2nd tells it to me) in the `msm8953-xiaomi-daisy.dts` because that seemed suspicious to me. But that also doesn’t solve it.

I am willing to test anything. I bought the phone for testing purposes. But am new to all this so can’t really help actively yet.

alikates, 2021-11-14:
Okay, i have cleaned up a bit the daisy branch, you should do a git fetch and git reset to the remote branch because i fixed the old commit and had to force-push. Also the goal here is to not get this lines:
```
[    2.219795] msm 1a00000.mdss: bound 1a01000.mdp (ops mdp5_ops)
[    2.226668] msm_dsi_manager_register: failed to register mipi dsi host for DSI 0: -517
```
They probably mean that the kernel is not finding the panel because the name is wrong or maybe the panel selection is not working properly.

Kiciuk, 2021-11-14:
Thats probe defer.
Its waiting for something.
backlight/supplies it also rekt
 cat /sys/kernel/debug/devices_deferred

apolemus, 2021-11-14:
That gives:
```
# cat /sys/kernel/debug/devices_deferred
1a01000.mdp	
c051000.sound-card	
200f000.spmi:pmic@0:temp-alarm@2400
```

apolemus, 2021-11-14:
I’m really sorry! Obviously I forgot to actually compile the panel driver. But at least the lk2nd issue got addressed. But the touchscreen probe still fails:
```
edt_ft5x06 1-0038: supply iovcc not found, using dummy regulator
...
edt_ft5x06 1-0038: touchscreen probe failed
```
Sorry again!

closed 2021-11-14 by apolemus