msm8953-mainline/linux #24: Upstream msm8953.dtsi

z3ntu, 2021-12-23:
We should aim to get at least basic parts of msm8953.dtsi upstream as it's quite well working but the file isn't upstream at all.

@SirSireesh Can I help with this, or do you already have a plan for this?

alikates, 2021-12-23:
Then we also need to upstream pmi8950.dtsi. It might need some cleanup to get it accepted as there are some device nodes that still haven't any matching driver upstream.

z3ntu, 2021-12-24:
I think also some of the macros defined there shouldn't be going upstream, I think it was just copy-pasted from downstream.

SirSireesh, 2021-12-24:
@z3ntu I've been trying to find time, but tbh its been hard as of late. If you can help that would be awesome!

z3ntu, 2021-12-24:
So prepare a .dtsi where all drivers are already upstream basically and follows upstream standards (node sorting, naming etc)? I can look into that. Really want to get 8953 more upstream :)

SirSireesh, 2021-12-26:
That sounds awesome, thanks! I'll try working on getting rpoc and IPA stuff upstream

Kiciuk, 2021-12-29:
@z3ntu i don't really think 8953 is ready to have upstreamed dts.
Do we really need to only provide base stuff upstream like pinctrl/gcc/usb nodes?
Otherwise there is not much more possible to be pushed, and worst situation is across dvfs(cpu-clk+CPR3)

z3ntu, 2021-12-30:
> i don't really think 8953 is ready to have upstreamed dts.

I did have to "throw away" a bunch of features as they're not upstream yet (and currently have some non-devicetree patches in my branch) but it's a good base feature set that can be improved on in the future. But the current situation definitely isn't good.

Also short status update from my side: I have a branch (as mentioned) that's looking rather good but I'm hitting some issues with USB that I need to debug, hopefully should be resolved in the next couple of days.

z3ntu, 2022-01-01:
First round is done, the regulators were silently not probing because I forgot to cherry-pick a commit, so USB never probed.

https://github.com/z3ntu/linux/commits/msm8953-upstream-1

As in user-visible features, only USB is supported, iommu and display and gpu need extra patches that I didn't want to bother with for now. If someone wants to try on their device I can prepare that device as well if you tell me.

alikates, 2022-01-01:
Have you looked at pmi8950 too? I see sdm632 devices don't need it but sdm625 ones do. Also the sdm632-pm8953 and msm8953-pm8953.dtsi files here only have a couple differences, so what do you think of merging them and making changes on a much smaller sdm632-specific file? I can try stuff on my xiaomi-daisy too.

z3ntu, 2022-01-02:
I've thrown away everything not relevant to my device, so no pmi8950, no msm8953-pm8953 and whatnot. 

So regarding the sdm632-pm8953 / msm8953-pm8953, we could make so devices need to include it like this
```
#include "msm8953-pm8953.dtsi"
#include "sdm632.dtsi"
```
and remove sdm632-pm8953.dtsi and keep msm8953-pm8953.dtsi basically how it is right now. Does that sound good?

alikates, 2022-01-02:
Yeah, looks good to me. I'm gonna be a couple days afk but then I can help to get the dts for sdm625 devices ready too.

Herrie82, 2022-01-04:
Is there somewhere a status of what works with which branch/fork for mainline? I'm about to try attempt to switch from Halium 9.0 to mainline for Mido, so would like to know which branch/fork to best use at the moment.

Kiciuk, 2022-01-04:
@Herrie82 Wiki status reflects support on the kernel which is packaged in pmaports so 5.13_rc6

[fd0f81dcec8023ea37d854c3d6d426ce41efb455](https://github.com/msm8953-mainline/linux/tree/fd0f81dcec8023ea37d854c3d6d426ce41efb455)

z3ntu, 2022-01-08:
v2: https://github.com/z3ntu/linux/commits/msm8953-upstream-2

Herrie82, 2022-01-08:
> (internal) v2: https://github.com/z3ntu/linux/commits/msm8953-upstream-2

Nice! What is the status in terms of hardware support here? Same as 5.13_rc6 on pmaports or things improved/got broken in the meanwhile?

Kiciuk, 2022-01-08:
> > (internal) v2: https://github.com/z3ntu/linux/commits/msm8953-upstream-2
> 
> Nice! What is the status in terms of hardware support here? Same as 5.13_rc6 on pmaports or things improved/got broken in the meanwhile?

This is tree which he made for sending patches to mainline its stripped AF pretty much booting to system with console accessible via SSH/USB.
Don;t expect much of new features to be added now.

z3ntu, 2022-01-09:
I apologize in advance for the annoying rebase this is causing though :wink: It definitely paves the way to get more upstream at some point though.

z3ntu, 2022-01-12:
Sent :tada: https://lore.kernel.org/linux-arm-msm/20220112194118.178026-1-luca@z3ntu.xyz/

SRIDHARDV, 2022-01-13:
Heeyyy... Congrats and happy to see MSM8953 in upstream 🎉

z3ntu, 2022-02-25:
Unless I missed something all patches from my initial series are now queued for kernel 5.18 :partying_face: I will double check linux-next on the weekend (once a new linux-next has been created)

z3ntu, 2022-03-15:
And the last patch has also finally landed in linux-next https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?id=a8f8cc6b39b7ee0dbaccbebd1268c9d3458ebf13

Should all be merged into the 5.18 merge window next week (or the week after). Closing!

@SirSireesh Might be worth to open new issues for the upstreaming status of individual components (e.g. remoteproc you seem to have worked on)