msm8953-mainline/linux #33: Xiaomi MI A2 Lite Daisy - touchpad does not work

qwertyuiopzxcvbnmlkjhgfdsa1, 2022-03-14:
On my phone display, which has display gt917d  is not working.

This is my personal issue for tracking the status , but everyone is of course free and welcome to help.

Last version of DTS, which could work, to be tested:

&i2c_3 {
	status = "okay";

	gt9xx@5d {
		compatible = "goodix,gt9xx";
		reg = <0x5d>;
		status = "okay";

		interrupt-parent = <&tlmm>;
		interrupts = <65 IRQ_TYPE_EDGE_FALLING>;

		reset-gpios = <&tlmm 64 0x0>;
		irq-gpios = <&tlmm 65 0x2008>;

		AVDD28-supply = <&pm8953_l10>;
		VDDIO-supply = <&pm8953_l6>;

		touchscreen-size-x = <1080>;
		touchscreen-size-y = <2280>;
	};

	focaltech@38 {
		compatible = "edt,edt-ft5406", "edt,edt-ft5x06";
		reg = <0x38>;

		interrupt-parent = <&tlmm>;
		interrupts = <65 IRQ_TYPE_EDGE_FALLING>;

		reset-gpios = <&tlmm 64 GPIO_ACTIVE_LOW>;

		vcc-supply = <&pm8953_l10>;

		touchscreen-size-x = <1080>;
		touchscreen-size-y = <2280>;
	};
};

M0Rf30, 2022-03-14:
* as reported [here](https://github.com/crdroidandroid/android_kernel_xiaomi_daisy/blob/b1edad1891ca8faec6c6c0f99063961f23b85c28/arch/arm64/boot/dts/qcom/sakura/sakura-qrd.dtsi#L112) should be goodix gf3208 
* IRQ_TYPE_EDGE_FALLING	= 0x00000002,  or 2 (include/dt-bindings/interrupt-controller/irq.h)

vldly, 2022-03-15:
1. Try to find [productinfo](https://github.com/goodix/goodix_gt9xx_public/blob/bcea65fee0b0fc9526ab817db87a30e0af23a4e1/kernel/drivers/input/touchscreen/gt9xx/gt9xx.c#L1114) in sysfs (probably in /sys/class/input/*/device).
2. No, person who contributed it has focaltech.
3. I wouldn't compare any magic values to downstream ones. Those depends on how driver uses it. 
Also [pinephone with gt917s uses IRQ_TYPE_LEVEL_HIGH](https://github.com/msm8953-mainline/linux/blob/e72845564bc120f1a16f9516ae1d616ea84ae519/arch/arm64/boot/dts/allwinner/sun50i-a64-pinephone.dtsi#L166). It might be right value for you too.
4. I suggest `loglevel=7 debug log_buf_len=10M` in cmdline and 
```
CONFIG_DYNAMIC_DEBUG=y
CONFIG_DEBUG_DRIVER=y
CONFIG_DEBUG_DEVRES=y
```
5. Author of the touchscreen driver chosen so. Check [dt-bindings](https://github.com/msm8953-mainline/linux/blob/e72845564bc120f1a16f9516ae1d616ea84ae519/Documentation/devicetree/bindings/input/touchscreen/goodix.yaml) and [driver source](https://github.com/msm8953-mainline/linux/blob/e72845564bc120f1a16f9516ae1d616ea84ae519/drivers/input/touchscreen/goodix.c) to know more.
6. Hopefully those are similar enough. But if it doesn't just work you can only rely on downstream source then. If you actually have 917d and it's similar gt917s you may want to add [chip_id entry](https://github.com/msm8953-mainline/linux/blob/e72845564bc120f1a16f9516ae1d616ea84ae519/drivers/input/touchscreen/goodix.c#L159) to the driver, because it defaults to gt9x_chip_data.

apolemus, 2022-03-17:
Assuming that downstream uses [this driver](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/daisy-q-oss/drivers/input/touchscreen/gt917d/gt9xx.c), 917D seems rather similar to 967 or something than to 917S, doesn’t it?

[REG_CONFIG_DATA](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/46a57878aa01f4e7bfc7ba0efd4a4ea02e7de557/drivers/input/touchscreen/gt917d/gt9xx.h#L308) is set to 0x8047 and not 0x8050 like for the 917S in the upstream driver.

And these [cfg groups](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/46a57878aa01f4e7bfc7ba0efd4a4ea02e7de557/drivers/input/touchscreen/gt917d/gt9xx.h#L197-L225) are 186 and 228 bits long – but not 240 like for the 917S. Why there are different cfg groups actually...

So I tried adding some chip_id entry combinations to mainline driver but wasn’t successful. In sysfs my 917D appears anyway but gives no output. I would like to test 4. but have no time for that atm.

vldly, 2022-03-18:
> Assuming that downstream uses [this driver](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/daisy-q-oss/drivers/input/touchscreen/gt917d/gt9xx.c), 917D seems rather similar to 967 or something than to 917S, doesn’t it?

You're right, i was judging just by model number.

> And these [cfg groups](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/46a57878aa01f4e7bfc7ba0efd4a4ea02e7de557/drivers/input/touchscreen/gt917d/gt9xx.h#L197-L225) are 186 and 228 bits long – but not 240 like for the 917S. Why there are different cfg groups actually...

No idea. But because of [CONFIG_OF being set](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/46a57878aa01f4e7bfc7ba0efd4a4ea02e7de557/drivers/input/touchscreen/gt917d/gt9xx.c#L970) it will parse it from [dt instead and it's 228 bytes there](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/46a57878aa01f4e7bfc7ba0efd4a4ea02e7de557/arch/arm/boot/dts/qcom/msm8953-qrd.dtsi#L146), just like on 967.

> In sysfs my 917D appears anyway but gives no output.

So it probed successfully? It might need the reset and config to work properly. You can easily create it by `cat /proc/device-tree/<path>/goodix,cfg-group0 > /sdcard/goodix_917D_cfg.bin` on android.

qwertyuiopzxcvbnmlkjhgfdsa1, 2022-03-18:
@vldly Do you think is it possible to get this info without rooted phone?

cat /proc/device-tree/<path>/goodix,cfg-group0 > /sdcard/goodix_917D_cfg.bin

apolemus, 2022-03-18:
> You can easily create it by cat /proc/device-tree/<path>/goodix,cfg-group0 > /sdcard/goodix_917D_cfg.bin on android.

I compiled the sequence you’ve given (and several others before) with [this script](https://patchwork.kernel.org/project/linux-input/patch/1438364532-27150-9-git-send-email-irina.tirdea@intel.com/), put it in /lib/firmware and got this:
```
xiaomi-daisy:~$ dmesg | grep Goodix
[   18.750445] Goodix-TS 3-005d: ID 917D, version: 501c
[   18.806238] input: Goodix Capacitive TouchScreen as /devices/platform/soc@0/78b7000.i2c/i2c-3/3-005d/input/input4
```
But it doesn’t work yet.

Before adding it, dmesg looks like this:
```
[   18.430649] Goodix-TS 3-005d: ID 917D, version: 501c
[   18.430757] Goodix-TS 3-005d: Direct firmware load for goodix_917D_cfg.bin failed with error -2
[   18.453243] input: Goodix Capacitive TouchScreen as /devices/platform/soc@0/78b7000.i2c/i2c-3/3-005d/input/input4
```

apolemus, 2022-03-18:
I had still gt917s as compatible in device tree. With gt967 I get (with and without firmware in /lib/firmware):

```
xiaomi-daisy:~$ dmesg | grep Goodix
[   18.049977] Goodix-TS 3-005d: Error reading 1 bytes from 0x8140: -6
[   18.078625] Goodix-TS 3-005d: ID \xeeM\x94\x8c, version: a9d9
[   18.101183] input: Goodix Capacitive TouchScreen as /devices/platform/soc@0/78b7000.i2c/i2c-3/3-005d/input/input3
```
So it’s not only.

qwertyuiopzxcvbnmlkjhgfdsa1, 2022-04-18:
Configuration for touchpad and other configurations are here. Look into firmware directory
https://github.com/qwertyuiopzxcvbnmlkjhgfdsa1/xiaomi-daisy-touchpad-config

autumnull, 2022-04-28:
I am also having this problem -- is there any further progress on a fix?

PetrBlahaPDI, 2022-04-28:
Last progress is here:  https://github.com/qwertyuiopzxcvbnmlkjhgfdsa1/linux/tree/msm8953-5.16

What is working there so far: 
-added 917d to driver
-when you copy configuration goodix_917D_cfg.bin  to /lib/firmware, it is loaded

There is remaining problem with IRQ...somehow the device gets the wrong IRQ(0).
Then in manage.c  there is line desc = irq_to_desc(irq);    where description should be get from device tree(in radix tree structure). But for IRQ=0 there is nothing in device tree.

Was asking for this, how to get the right IRQ -> this is set automatically, not in dts. Somebody replied, that the irq number is got from parent device. Not sure what do to now, tried with irq-flags = <2>; al well, does not help.

So there is progress, but i am stuck, not having knowledge, how to proceed next. So abandoning this task for now.

qwertyuiopzxcvbnmlkjhgfdsa1, 2022-04-28:
Last progress is here:  https://github.com/qwertyuiopzxcvbnmlkjhgfdsa1/linux/tree/msm8953-5.16

What is working there so far: 
-added 917d to driver
-when you copy configuration goodix_917D_cfg.bin  to /lib/firmware, it is loaded

There is remaining problem with IRQ...somehow the device gets the wrong IRQ(0).
Then in manage.c  there is line desc = irq_to_desc(irq);    where description should be get from device tree(in radix tree structure). But for IRQ=0 there is nothing in device tree.

Was asking for this, how to get the right IRQ -> this is set automatically, there is probe function which set up many settings for drivers, i think setting anything in dts wont help. Somebody replied, that the irq number is got from parent device. Not sure what do to now, tried with irq-flags = <2>; al well, does not help.

So there is progress, but i am stuck, not having knowledge, how to proceed next. So abandoning this task for now.

psyhlo, 2022-07-15:
Hello, how can i help with this problem i have the same phone but not sure which touchpad?
i have install postmarketos and boots but no touch response so far
how can i check which model touch panel i have?

apolemus, 2022-07-15:
You can get some information out of dmesg (e.g. `dmesg | grep Goodix`). Also the goodix gt917d shows up in sysfs. But like above discussed, this works but there is no output anyway.

psyhlo, 2022-07-16:
i have gt917s
found it in /sys/devices/platform/soc@0/78b7000.i2c/i2c-3/3-005d/name
if any want some info from my device, just tell
i'll try something on my own

> xiaomi-daisy:~$ dmesg | grep Goodix
> [   18.279106] Goodix-TS 3-005d: ID 917D, version: 501c
> [   18.279269] Goodix-TS 3-005d: Direct firmware load for goodix_917D_cfg.bin failed with error -2
> [   18.301759] input: Goodix Capacitive TouchScreen as /devices/platform/soc@0/78b7000.i2c/i2c-3/3-005d/input/input4

psyhlo, 2022-07-20:
i found this driver for kernel 3.18 https://github.com/anjum51/android_kernel_xiaomi_onclite/tree/884c3832d9e115fcd45cd15bda25f5edaeac29f2/drivers/input/touchscreen
has someone tried to port it?

apolemus, 2022-07-20:
No, that will probably make different problems. It’s highly likely that the Goodix touchscreen driver in mainline will work also for this gt917d. Probably we just have to adjust the daisy dts for getting the irq correctly working. But somehow everyone with this problem has not the knowledge yet how to do that. ;)

alikates, 2022-12-12:
Hi, any update with this?

Mr-Yuling, 2023-04-03:
> On my phone display, which has display ILITEK ili7807-fdhplus 1080x2280 video and probably touchpad gt917d is not working.
> 
> This is my personal issue for tracking the status , but everyone is of course free and welcome to help. Inspiration here: https://github.com/goodix/goodix_gt9xx_public
> 
> https://github.com/LineageOS/android_kernel_xiaomi_msm8953/blob/lineage-18.1/arch/arm64/boot/dts/qcom/apq8053-lite-dragon-v2.2.dtsi
> 
> Daisy config here: https://github.com/qwertyuiopzxcvbnmlkjhgfdsa1/xiaomi-daisy-touchpad-config
> 
> UPDATE2: Configuration is correctly applied, remaining issue is with IRQ. There is a function which tries to find description in devices radix tree structure by irq 0, but there is no description, thus it fails. Debugging why it happend.
> 
> UPDATE3:IRQ=0 it could be parent irq, nobody knows much
> 
> ARCHIVE ###Old Open questions: 1, Is my touchpad really goodix gt917d? 2, Is gt917s, which is base of this work actually working for anyone? I doubt that those two original lines work, there should be surely different values: reset-gpios = <&tlmm 64 GPIO_ACTIVE_HIGH>; irq-gpios = <&tlmm 65 GPIO_ACTIVE_HIGH>; 3, How to get know what value holds magic strings like "IRQ_TYPE_EDGE_FALLING"? 4, Some debuging - howto get more debug on boot? 5, Why AVDD28 is uppercase? And how original author know what names to use? 6, Datasheet for gt917d are not to be found. There are only for gt917s. Strange.

I use redmi 6pro (the same as Mi A2 Lite),also has display ili7807-fdhplus,but it can not display.

qwertyuiopzxcvbnmlkjhgfdsa1, 2023-04-03:
Do not have phone any longer, but found driver, which may be perhaps inspiration
https://gitlab.com/postmarketOS/pmaports/-/merge_requests/3963

Mr-Yuling, 2023-04-04:
> Do not have phone any longer, but found driver, which may be perhaps inspiration https://gitlab.com/postmarketOS/pmaports/-/merge_requests/3963

I have resolved the display,also having this problem.touchpad can not use.
`[  102.680660] Goodix-TS 2-005d: ID 917D, version: 501c
[  102.759884] input: Goodix Capacitive TouchScreen as /devices/platform/soc@0/78b7000.i2c/i2c-2/2-005d/input/input5
`

ntonnaett, 2023-10-21:
If you test #128 make sure to also apply the lk2nd patch that it depends on.

tphlrussia, 2023-12-02:
Any further progress ?

barni2000, 2024-01-02:
Already merged this should be closed.

qwertyuiopzxcvbnmlkjhgfdsa1, 2024-01-02:
fixed

PR #128 (2023-12-30): Add Goodix GT917D touchscreen support for Xiaomi Daisy 
Depends on https://github.com/msm8953-mainline/lk2nd/pull/57 for enabling the correct touchscreen according to the display.

Fixes #33
closed 2024-01-02 by qwertyuiopzxcvbnmlkjhgfdsa1