msm8953-mainline/linux #38: tryed to compile with xiaomi-mido panels and got this errors

privatgt, 2022-05-25:
In file included from drivers/gpu/drm/panel/panels-xiaomi-mido.c:8:
./include/linux/module.h:131:49: error: redefinition of '__inittest'
  131 |         static inline initcall_t __maybe_unused __inittest(void)                \
      |                                                 ^~~~~~~~~~
./include/linux/device/driver.h:263:1: note: in expansion of macro 'module_init'
  263 | module_init(__driver##_init); \
      | ^~~~~~~~~~~
./include/drm/drm_mipi_dsi.h:330:9: note: in expansion of macro 'module_driver'
  330 |         module_driver(__mipi_dsi_driver, mipi_dsi_driver_register, \
      |         ^~~~~~~~~~~~~
drivers/gpu/drm/panel/panel-mipi-dsi-common.h:24:9: note: in expansion of macro 'module_mipi_dsi_driver'
   24 |         module_mipi_dsi_driver(_prefix##_driver)
      |         ^~~~~~~~~~~~~~~~~~~~~~
drivers/gpu/drm/panel/panels-xiaomi-mido.c:506:1: note: in expansion of macro 'MIPI_DSI_PANEL_DRIVER'
  506 | MIPI_DSI_PANEL_DRIVER(otm1911_fhd, "otm1911-fhd", "xiaomi,otm1911");
      | ^~~~~~~~~~~~~~~~~~~~~
./include/linux/module.h:131:49: note: previous definition of '__inittest' with type 'int (*(void))(void)'
  131 |         static inline initcall_t __maybe_unused __inittest(void)                \
      |                                                 ^~~~~~~~~~
./include/linux/device/driver.h:263:1: note: in expansion of macro 'module_init'
  263 | module_init(__driver##_init); \
      | ^~~~~~~~~~~
./include/drm/drm_mipi_dsi.h:330:9: note: in expansion of macro 'module_driver'
  330 |         module_driver(__mipi_dsi_driver, mipi_dsi_driver_register, \
      |         ^~~~~~~~~~~~~
drivers/gpu/drm/panel/panel-mipi-dsi-common.h:24:9: note: in expansion of macro 'module_mipi_dsi_driver'
   24 |         module_mipi_dsi_driver(_prefix##_driver)
      |         ^~~~~~~~~~~~~~~~~~~~~~
drivers/gpu/drm/panel/panels-xiaomi-mido.c:505:1: note: in expansion of macro 'MIPI_DSI_PANEL_DRIVER'
  505 | MIPI_DSI_PANEL_DRIVER(r63350_ebbg_fhd, "r63350-ebbg-fhd", "xiaomi,ebbg-r63350");
      | ^~~~~~~~~~~~~~~~~~~~~
./include/linux/module.h:133:13: error: redefinition of 'init_module'
  133 |         int init_module(void) __copy(initfn)                    \
      |             ^~~~~~~~~~~
./include/linux/device/driver.h:263:1: note: in expansion of macro 'module_init'
  263 | module_init(__driver##_init); \
      | ^~~~~~~~~~~
./include/drm/drm_mipi_dsi.h:330:9: note: in expansion of macro 'module_driver'
  330 |         module_driver(__mipi_dsi_driver, mipi_dsi_driver_register, \
      |         ^~~~~~~~~~~~~
drivers/gpu/drm/panel/panel-mipi-dsi-common.h:24:9: note: in expansion of macro 'module_mipi_dsi_driver'
   24 |         module_mipi_dsi_driver(_prefix##_driver)
      |         ^~~~~~~~~~~~~~~~~~~~~~
drivers/gpu/drm/panel/panels-xiaomi-mido.c:506:1: note: in expansion of macro 'MIPI_DSI_PANEL_DRIVER'
  506 | MIPI_DSI_PANEL_DRIVER(otm1911_fhd, "otm1911-fhd", "xiaomi,otm1911");
      | ^~~~~~~~~~~~~~~~~~~~~
./include/linux/module.h:133:13: note: previous definition of 'init_module' with type 'int(void)'
  133 |         int init_module(void) __copy(initfn)                    \
      |             ^~~~~~~~~~~
./include/linux/device/driver.h:263:1: note: in expansion of macro 'module_init'
  263 | module_init(__driver##_init); \
      | ^~~~~~~~~~~
./include/drm/drm_mipi_dsi.h:330:9: note: in expansion of macro 'module_driver'
  330 |         module_driver(__mipi_dsi_driver, mipi_dsi_driver_register, \
      |         ^~~~~~~~~~~~~
drivers/gpu/drm/panel/panel-mipi-dsi-common.h:24:9: note: in expansion of macro 'module_mipi_dsi_driver'
   24 |         module_mipi_dsi_driver(_prefix##_driver)
      |         ^~~~~~~~~~~~~~~~~~~~~~
drivers/gpu/drm/panel/panels-xiaomi-mido.c:505:1: note: in expansion of macro 'MIPI_DSI_PANEL_DRIVER'
  505 | MIPI_DSI_PANEL_DRIVER(r63350_ebbg_fhd, "r63350-ebbg-fhd", "xiaomi,ebbg-r63350");
      | ^~~~~~~~~~~~~~~~~~~~~
./include/linux/module.h:139:49: error: redefinition of '__exittest'
  139 |         static inline exitcall_t __maybe_unused __exittest(void)                \
      |                                                 ^~~~~~~~~~
./include/linux/device/driver.h:268:1: note: in expansion of macro 'module_exit'
  268 | module_exit(__driver##_exit);
      | ^~~~~~~~~~~
./include/drm/drm_mipi_dsi.h:330:9: note: in expansion of macro 'module_driver'
  330 |         module_driver(__mipi_dsi_driver, mipi_dsi_driver_register, \
      |         ^~~~~~~~~~~~~
[... 385 lines cut ...]
  268 | module_exit(__driver##_exit);
      | ^~~~~~~~~~~
./include/drm/drm_mipi_dsi.h:330:9: note: in expansion of macro 'module_driver'
  330 |         module_driver(__mipi_dsi_driver, mipi_dsi_driver_register, \
      |         ^~~~~~~~~~~~~
drivers/gpu/drm/panel/panel-mipi-dsi-common.h:24:9: note: in expansion of macro 'module_mipi_dsi_driver'
   24 |         module_mipi_dsi_driver(_prefix##_driver)
      |         ^~~~~~~~~~~~~~~~~~~~~~
drivers/gpu/drm/panel/panels-xiaomi-mido.c:509:1: note: in expansion of macro 'MIPI_DSI_PANEL_DRIVER'
  509 | MIPI_DSI_PANEL_DRIVER(ili9885_boe_fhd, "ili9885-boe-fhd", "xiaomi,boe-ili9885");
      | ^~~~~~~~~~~~~~~~~~~~~
./include/linux/module.h:141:14: note: previous definition of 'cleanup_module' with type 'void(void)'
  141 |         void cleanup_module(void) __copy(exitfn)                \
      |              ^~~~~~~~~~~~~~
./include/linux/device/driver.h:268:1: note: in expansion of macro 'module_exit'
  268 | module_exit(__driver##_exit);
      | ^~~~~~~~~~~
./include/drm/drm_mipi_dsi.h:330:9: note: in expansion of macro 'module_driver'
  330 |         module_driver(__mipi_dsi_driver, mipi_dsi_driver_register, \
      |         ^~~~~~~~~~~~~
drivers/gpu/drm/panel/panel-mipi-dsi-common.h:24:9: note: in expansion of macro 'module_mipi_dsi_driver'
   24 |         module_mipi_dsi_driver(_prefix##_driver)
      |         ^~~~~~~~~~~~~~~~~~~~~~
drivers/gpu/drm/panel/panels-xiaomi-mido.c:505:1: note: in expansion of macro 'MIPI_DSI_PANEL_DRIVER'
  505 | MIPI_DSI_PANEL_DRIVER(r63350_ebbg_fhd, "r63350-ebbg-fhd", "xiaomi,ebbg-r63350");

Kiciuk, 2022-05-28:
Mido panels broken till we bump kernel to 5.18

privatgt, 2022-06-01:
> Mido panels broken till we bump kernel to 5.18

I just compiled only with my display panel and it worked

`-MIPI_DSI_PANEL_DRIVER(r63350_ebbg_fhd, "r63350-ebbg-fhd", "xiaomi,ebbg-r63350");
-MIPI_DSI_PANEL_DRIVER(otm1911_fhd, "otm1911-fhd", "xiaomi,otm1911");
-MIPI_DSI_PANEL_DRIVER(nt35596_tianma_fhd, "nt35596-tianma-fhd", "xiaomi,tianma-nt35596");
 MIPI_DSI_PANEL_DRIVER(nt35532_fhd, "nt35532-fhd", "xiaomi,nt35532");
-MIPI_DSI_PANEL_DRIVER(ili9885_boe_fhd, "ili9885-boe-fhd", "xiaomi,boe-ili9885");`

SRIDHARDV, 2022-11-12:
> Mido panels broken till we bump kernel to 5.18

Hi Any updates in kernel 5.18 or in 6.0.

z3ntu, 2022-12-12:
Should work again, let's close unless it's still broken on 6.0.10 branch

closed 2022-12-12 by z3ntu