msm8953-mainline/linux #53: Update to linux 6.0

alikates, 2022-10-13:
I just pushed a [WIP branch](https://github.com/msm8953-mainline/linux/tree/msm8953-6.0-wip) based on linux 6.0. Here I want to make something like a ToDo list to track the status of the update.

Also I'm thinking on squashing all the typo fixes to clean up the history.

Regarding the panel drivers, maybe we should do the same as on msm8916-mainline and automatically add all panel drivers with lmdpdg. What do you think?

Please tell me what else to add:
- [ ] Clean up device DTs.
- [ ] Further clean up all SoC commits.
- [ ] Squash all typo and fix commits.
- [ ] Review the IOMMU commits.
- [ ] Figure out what happened with the CPU and GPU frequency scaling (OPP tables) and fix it.
- [ ] Get IPA working (maybe? I have a WIP rebase elsewhere and it almost works).
- [ ] Enable the charger. (I'm not sure about this because we can't reliably test it and it's still an RFC...)
- [ ] Update configs

It can already be tested on phones although it's probably a bit slow because of the OPP stuff.

Tested on:
- [ ] mido
- [ ] daisy
- [ ] tissot
- [ ] ocean
- [ ] vince
- [ ] fp3
- [ ] a6plte

M0Rf30, 2022-10-13:
Great initiative. 
Can I add:
- usage of labibb implementation (already mainlined) when occurred for panels
(take it as humble suggestion)

z3ntu, 2022-10-13:
If you haven't already, please add https://github.com/msm8953-mainline/linux/pull/43

xnstet, 2022-10-14:
i had vince. but after the new update (>20 september 2022) i got many troubles reinstalling pmos. mainline, no display at all ui i tried. now the only one working ui is sxmo-de-dwm on downstream and when i try to watch youtube at the browser, crashed. i hope 6.0 is the answer next. btw how to git clone this 6.0wip repository? i want to compile-install-flash immediately. thanks for the hard work, devs. thanks

alikates, 2022-10-16:
Okay, I'm cherry picking all of that. Also what do you think of squashing all DT commits?

M0Rf30, 2022-10-17:
Yeah, for me it's ok

alikates, 2022-10-21:
Sorry for that. I just added the labibb, hx83112b, and fp3 touchscreen stuff

z3ntu, 2022-10-22:
At least this commit from 5.18.3 branch is missing in 6.0 branch: d958d986679cfd2da3c129c6fcb33de91956b412

CC @Ultracoolguy

alikates, 2022-10-22:
Yep i merged it yesterday and forgot to cherry pick it

alikates, 2022-10-22:
I'm going to try somainline CPR4 driver and see if it changes anything. Current driver looks like is stuck trying to probe

M0Rf30, 2022-10-23:
Can you replicate this issue at build time:
```
sound/soc/codecs/tfa9872.c:120:10: error: 'const struct snd_soc_component_driver' has no member named 'non_legacy_dai_naming'; did you mean 'legacy_dai_naming'?
  120 |         .non_legacy_dai_naming  = 1,
      |          ^~~~~~~~~~~~~~~~~~~~~
      |          legacy_dai_naming
make[3]: *** [scripts/Makefile.build:249: sound/soc/codecs/tfa9872.o] Error 1
make[3]: *** Waiting for unfinished jobs....
```

M0Rf30, 2022-10-23:
Can you replicate this?
```
  DTC     arch/arm64/boot/dts/qcom/sdm632-motorola-ocean.dtb
../arch/arm64/boot/dts/qcom/msm8953.dtsi:2292.4-27: Warning (interrupts_property): /soc@0/i2c@78b7000/touchscreen@20:#interrupt-cells: size is (8), expected multiple of 12
  DTC     arch/arm64/boot/dts/qcom/sdm636-sony-xperia-ganges-mermaid.dtb
```

z3ntu, 2022-10-24:
@M0Rf30 See e.g. https://github.com/torvalds/linux/commit/a524837ddd11bc20ec59d033d0260707cfa3cb99

Ultracoolguy, 2022-10-25:
Cherry picking the commit z3ntu mentioned and pushing a commit that fixes the tfa9872 error makes 6.0 to function on ocean. However there's an issue with the panel I'm working on that seems dt related.

Ultracoolguy, 2022-10-25:
I fixed that too.

z3ntu, 2022-11-01:
```
[    1.045448] Error: Driver 'panel-otm1911-fhd' is already registered, aborting...
```
Seems there's a duplicate driver
```
drivers/gpu/drm/panel/msm8953-generated/panel-mdss-otm1911-fhd.c:               .name = "panel-otm1911-fhd",
drivers/gpu/drm/panel/msm8953-generated/panel-xiaomi-otm1911.c:         .name = "panel-otm1911-fhd",
```

Additionally https://github.com/Kiciuk/linux-panel-drivers should be moved into msm8953-mainline namespace.

I'll also send a PR to this repo for panel driver because that was dropped during the rebase.

z3ntu, 2022-11-02:
Tested 6.0 branch on FP3 properly now, I sometimes get a reboot briefly after bootup, nothing on serial log about it though.

Apart from that, adsp, wifi, bluetooth, touchscreen all work fine. Venus not sure, but also don't think I tested in the past, modem also maybe not but I also don't recall that ever working properly

M0Rf30, 2022-11-08:
> Panel seems to be locked to 30fps now which makes everything quite laggy. Checked with kmscube and glxgears, both hover at or under 30fps for some reason

Could be related to LABIBB regulator usage?

fekz115, 2022-11-08:
> Panel seems to be locked to 30fps now which makes everything quite laggy. Checked with kmscube and glxgears, both hover at or under 30fps for some reason

TBH, did you checked with both xorg and wayland DEs?

z3ntu, 2022-11-09:
glxgears was on Phosh or Plasma Mobile, I don't recall. kmscube uses neither.

fekz115, 2022-11-09:
> glxgears was on Phosh or Plasma Mobile, I don't recall. kmscube uses neither.

Try on xfce, please

z3ntu, 2022-11-28:
Perhaps it's a good idea to make sure functionality 6.0 works and defer the cleanup and whatever other tasks to either a future rebase or put them later onto 6.0 branch? Otherwise I think this is going to take forever.

alikates, 2022-11-28:
Yes, I also didnt have much time to work on it. Also i tried to rebase everything on top of 6.1-rc7. Do we wait for the final release or just use 6.0?

Btw, what about OPPs and scaling? Is it worth trying to fix it for this release?

z3ntu, 2022-11-28:
Latest 6.0.x would be good, so right now 6.0.10.

OPP/scaling, not sure. Does rebasing the 5.18(?) commits don't work?

alikates, 2022-12-06:
I rebased everything on top of 6.0.10. I think we should stick with that version so please open PRs against this branch
https://github.com/msm8953-mainline/linux/tree/msm8953-6.0.10

I reordered dts and config commits so its easier to cleanup and squash and i replaced the IOMMU commits with the v2 from the mailing list.

alikates, 2022-12-06:
For using the charger it's just setting its status to "okay" in the phone DTS and enabling the config

M0Rf30, 2022-12-13:
I updated this MR https://gitlab.com/postmarketOS/pmaports/-/merge_requests/3697 to latest changes from this branch

alikates, 2022-12-26:
I think we can reuse this issue as a ToDo list for next releases

z3ntu, 2022-12-26:
But let's close this issue since 6.0.10 is merged in pmaports. If there's anything missing please open a separate one.

closed 2022-12-26 by z3ntu