msm8953-mainline/linux #55: Upstream GPU support

z3ntu, 2022-10-18:
Required getting rid of some hacks like 3bc402701599342e2780e13d68f4ffcf378c119f and probably some IOMMU things.

@vldly Can you share any details why these hacks exist right now and a possible way to not need them anymore?

alikates, 2023-03-28:
I think this hack is just so Mesa treats the GPU as supported. We just need to add A506 here https://gitlab.freedesktop.org/mesa/mesa/-/blob/main/src/freedreno/common/freedreno_devices.py#L180

M0Rf30, 2023-03-28:
> You'll _probably_ also need something like [sdm660-mainline@43eb92b](https://github.com/sdm660-mainline/linux/commit/43eb92b8babf14da947f9056e11205961a9066c6) (but I'm not 100% sure)?

MODULE_FIRMWARE("qcom/a506_zap.b00");
MODULE_FIRMWARE("qcom/a506_zap.b01");
MODULE_FIRMWARE("qcom/a506_zap.b02");
MODULE_FIRMWARE("qcom/a506_zap.mdt");

z3ntu, 2023-03-30:
FTR from matrix: https://matrix.to/#/!znmHDDTlttuCQhhnWN:matrix.org/$VEBnGhxv-tJrpZEAq17C3eeCGK5QQIX7wdOfeI6b9L8?via=z3ntu.xyz&via=matrix.org

> **vladly**
> bare minimum for GPU to work is:
> 
> just GPU IOMMU and GPU nodes.
> `iommu/qcom: defer probe until scm is available`
> `iommu/qcom: restore IOMMU state if needed`
> `iommu/qcom: Use the asid read from device-tree if specified` or `iommu/qcom: Index contexts by asid number to allow asid 0`
> 
> **alikates (he/they)**
> The last one was already merged
> Author: AngeloGioacchino Del Regno [angelogioacchino.delregno@collabora.com](mailto:angelogioacchino.delregno@collabora.com)
> Date: Tue Nov 15 11:11:20 2022 +0100
> 
> iommu/qcom: Index contexts by asid number to allow asid 0

z3ntu, 2023-04-23:
Seems IOMMU patches are still pending, so we're blocked on the kernel part on that. 

But in the meantime I've submitted mesa MR which makes GPU work without the A530 hack linked in the top message https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/22647

z3ntu, 2023-04-25:
A506 patch is also now present in Alpine edge: https://gitlab.alpinelinux.org/alpine/aports/-/merge_requests/46345

So we can start trying to drop the 3c181c25e8355d641a711dc5da4ab35b1b5a64fb commit I believe, worked fine in my testing.

spasswolf, 2023-06-06:
I just cherry-picked the iommu patches (starting with 8f2410192ef7255e0bc2cec6b29d33412002854) on top of kernel.org linux-next-20230602. To get drm to initialize one also needs drivers/interconnect/qcom/msm8953.c, otherwise the booting process seems to crash (on fairphone-fp3). Now only the display fails to initialize:
```
[    3.179743] msm_mdp 1a01000.display-controller: bound 1c00000.gpu (ops 0xffff800008d2a190)
[    3.179941] msm_mdp 1a01000.display-controller: [drm:mdp5_kms_init] MDP5 version v1.16
[    3.205728] [drm] Initialized msm 1.10.0 20130625 for 1a01000.display-controller on minor 0
[    3.206002] msm_mdp 1a01000.display-controller: [drm:adreno_request_fw] loaded qcom/a530_pm4.fw from legacy location
[    3.213653] msm_mdp 1a01000.display-controller: [drm:adreno_request_fw] loaded qcom/a530_pfp.fw from legacy location
[    3.307622] msm_mdp 1a01000.display-controller: pp done time out, lm=0
[    3.340772] panel-djn-hx83112b 1a94000.dsi.0: Failed to initialize panel: -22

```
Almost there ...

spasswolf, 2023-06-07:
These are the patches that make the gpu and display work for the fairphone-fp3 for linux-next-20230602 from kernel.org. I've not yet tried how well these work with the current linux-next/master.
[0001-iommu-qcom-Use-the-asid-read-from-device-tree-if-spe.patch.txt](https://github.com/msm8953-mainline/linux/files/11678689/0001-iommu-qcom-Use-the-asid-read-from-device-tree-if-spe.patch.txt)
[0002-iommu-qcom-Properly-reset-the-IOMMU-context.patch.txt](https://github.com/msm8953-mainline/linux/files/11678693/0002-iommu-qcom-Properly-reset-the-IOMMU-context.patch.txt)
[0003-iommu-qcom-Index-contexts-by-asid-number-to-allow-as.patch.txt](https://github.com/msm8953-mainline/linux/files/11678694/0003-iommu-qcom-Index-contexts-by-asid-number-to-allow-as.patch.txt)
[0004-iommu-qcom-Add-support-for-QSMMUv2-and-QSMMU-500-sec.patch.txt](https://github.com/msm8953-mainline/linux/files/11678696/0004-iommu-qcom-Add-support-for-QSMMUv2-and-QSMMU-500-sec.patch.txt)
[0005-iommu-qcom-Add-support-for-AArch64-IOMMU-pagetables.patch.txt](https://github.com/msm8953-mainline/linux/files/11678700/0005-iommu-qcom-Add-support-for-AArch64-IOMMU-pagetables.patch.txt)
[0006-fixup-iommu-qcom-Add-support-for-AArch64-IOMMU-paget.patch.txt](https://github.com/msm8953-mainline/linux/files/11678703/0006-fixup-iommu-qcom-Add-support-for-AArch64-IOMMU-paget.patch.txt)
[0007-iommu-qcom-avoid-bouncing-power-domain.patch.txt](https://github.com/msm8953-mainline/linux/files/11678704/0007-iommu-qcom-avoid-bouncing-power-domain.patch.txt)
[0008-iommu-qcom-implement-adreno_priv-interface.patch.txt](https://github.com/msm8953-mainline/linux/files/11678706/0008-iommu-qcom-implement-adreno_priv-interface.patch.txt)
[0009-iommu-qcom-emulate-IDENTITY-domain-for-display-durin.patch.txt](https://github.com/msm8953-mainline/linux/files/11678707/0009-iommu-qcom-emulate-IDENTITY-domain-for-display-durin.patch.txt)
[0010-iommu-qcom-restore-IOMMU-state-if-needed.patch.txt](https://github.com/msm8953-mainline/linux/files/11678708/0010-iommu-qcom-restore-IOMMU-state-if-needed.patch.txt)
[0011-iommu-qcom-defer-probe-until-scm-is-available.patch.txt](https://github.com/msm8953-mainline/linux/files/11678709/0011-iommu-qcom-defer-probe-until-scm-is-available.patch.txt)
[0012-MSM8953-drm-panel-Generate-using-linux-mdss-dsi-pane.patch.txt](https://github.com/msm8953-mainline/linux/files/11678711/0012-MSM8953-drm-panel-Generate-using-linux-mdss-dsi-pane.patch.txt)
[0013-squash-MSM8953-drm-panel-Generate-using-linux-mdss-d.patch.txt](https://github.com/msm8953-mainline/linux/files/11678712/0013-squash-MSM8953-drm-panel-Generate-using-linux-mdss-d.patch.txt)
[0014-drm-panel-fairphone-fp3-hx83112b-Swab-brightness-val.patch.txt](https://github.com/msm8953-mainline/linux/files/11678713/0014-drm-panel-fairphone-fp3-hx83112b-Swab-brightness-val.patch.txt)
[0015-squash-MSM8953-drm-panel-Generate-using-linux-mdss-d.patch.txt](https://github.com/msm8953-mainline/linux/files/11678715/0015-squash-MSM8953-drm-panel-Generate-using-linux-mdss-d.patch.txt)
[0016-arm64-dts-qcom-msm8953-add-IOMMUs.patch.txt](https://github.com/msm8953-mainline/linux/files/11678716/0016-arm64-dts-qcom-msm8953-add-IOMMUs.patch.txt)
[0017-arm64-dts-qcom-msm8953-add-GPU.patch.txt](https://github.com/msm8953-mainline/linux/files/11678718/0017-arm64-dts-qcom-msm8953-add-GPU.patch.txt)
[0018-arm64-dts-qcom-msm8953-add-GPU-operating-points.patch.txt](https://github.com/msm8953-mainline/linux/files/11678719/0018-arm64-dts-qcom-msm8953-add-GPU-operating-points.patch.txt)
[0019-arm64-dts-qcom-msm8953-enable-cpu-frequency-scaling.patch.txt](https://github.com/msm8953-mainline/linux/files/11678721/0019-arm64-dts-qcom-msm8953-enable-cpu-frequency-scaling.patch.txt)
[0020-arm64-dts-qcom-msm8953-add-cci-adapter.patch.txt](https://github.com/msm8953-mainline/linux/files/11678722/0020-arm64-dts-qcom-msm8953-add-cci-adapter.patch.txt)
[0021-arm64-dts-qcom-msm8953-add-camss-subsystem.patch.txt](https://github.com/msm8953-mainline/linux/files/11678725/0021-arm64-dts-qcom-msm8953-add-camss-subsystem.patch.txt)
[0022-arm64-dts-qcom-msm8953-add-cpu-idle-states.patch.txt](https://github.com/msm8953-mainline/linux/files/11678726/0022-arm64-dts-qcom-msm8953-add-cpu-idle-states.patch.txt)
[0023-arm64-dts-qcom-sdm632-enable-cpu-frequency-scaling.patch.txt](https://github.com/msm8953-mainline/linux/files/11678727/0023-arm64-dts-qcom-sdm632-enable-cpu-frequency-scaling.patch.txt)
[0024-arm64-dts-qcom-msm8953-add-modem-remoteproc.patch.txt](https://github.com/msm8953-mainline/linux/files/11678728/0024-arm64-dts-qcom-msm8953-add-modem-remoteproc.patch.txt)
[0025-arm64-dts-qcom-sdm632-fairphone-fp3-add-display.patch.txt](https://github.com/msm8953-mainline/linux/files/11678729/0025-arm64-dts-qcom-sdm632-fairphone-fp3-add-display.patch.txt)
[0026-fixup-arm64-dts-qcom-sdm632-fairphone-fp3-add-displa.patch.txt](https://github.com/msm8953-mainline/linux/files/11678731/0026-fixup-arm64-dts-qcom-sdm632-fairphone-fp3-add-displa.patch.txt)
[0027-HACK-drm-msm-a5xx-present-A506-as-A530.patch.txt](https://github.com/msm8953-mainline/linux/files/11678732/0027-HACK-drm-msm-a5xx-present-A506-as-A530.patch.txt)
[0028-clk-qcom-gcc-msm8953-don-t-cache-rate-for-dsi-clocks.patch.txt](https://github.com/msm8953-mainline/linux/files/11678733/0028-clk-qcom-gcc-msm8953-don-t-cache-rate-for-dsi-clocks.patch.txt)
[... cut ...]
[0035-arm64-dts-qcom-msm8953-add-interconnect.patch.txt](https://github.com/msm8953-mainline/linux/files/11678746/0035-arm64-dts-qcom-msm8953-add-interconnect.patch.txt)
[0036-interconnect-qcom-Add-MSM8953-interconnect-provider-.patch.txt](https://github.com/msm8953-mainline/linux/files/11678747/0036-interconnect-qcom-Add-MSM8953-interconnect-provider-.patch.txt)
[0037-Revert-drm-msm-dsi-Stop-unconditionally-powering-up-.patch.txt](https://github.com/msm8953-mainline/linux/files/11678748/0037-Revert-drm-msm-dsi-Stop-unconditionally-powering-up-.patch.txt)

spasswolf, 2023-06-19:
I was able to trim it down a little. These are the patches necessary to have working GPU and display on the fairphone-fp3, patches are on top of linux-next-20230616 from kernel.org (commit f7efed9f38f886edb450041b82a6f15d663c98f8):
[0001-iommu-qcom-Use-the-asid-read-from-device-tree-if-spe.patch.txt](https://github.com/msm8953-mainline/linux/files/11789356/0001-iommu-qcom-Use-the-asid-read-from-device-tree-if-spe.patch.txt)
[0002-iommu-qcom-Properly-reset-the-IOMMU-context.patch.txt](https://github.com/msm8953-mainline/linux/files/11789359/0002-iommu-qcom-Properly-reset-the-IOMMU-context.patch.txt)
[0003-iommu-qcom-Index-contexts-by-asid-number-to-allow-as.patch.txt](https://github.com/msm8953-mainline/linux/files/11789364/0003-iommu-qcom-Index-contexts-by-asid-number-to-allow-as.patch.txt)
[0004-iommu-qcom-Add-support-for-QSMMUv2-and-QSMMU-500-sec.patch.txt](https://github.com/msm8953-mainline/linux/files/11789366/0004-iommu-qcom-Add-support-for-QSMMUv2-and-QSMMU-500-sec.patch.txt)
[0005-iommu-qcom-Add-support-for-AArch64-IOMMU-pagetables.patch.txt](https://github.com/msm8953-mainline/linux/files/11789367/0005-iommu-qcom-Add-support-for-AArch64-IOMMU-pagetables.patch.txt)
[0006-fixup-iommu-qcom-Add-support-for-AArch64-IOMMU-paget.patch.txt](https://github.com/msm8953-mainline/linux/files/11789368/0006-fixup-iommu-qcom-Add-support-for-AArch64-IOMMU-paget.patch.txt)
[0007-iommu-qcom-avoid-bouncing-power-domain.patch.txt](https://github.com/msm8953-mainline/linux/files/11789371/0007-iommu-qcom-avoid-bouncing-power-domain.patch.txt)
[0008-iommu-qcom-implement-adreno_priv-interface.patch.txt](https://github.com/msm8953-mainline/linux/files/11789373/0008-iommu-qcom-implement-adreno_priv-interface.patch.txt)
[0009-iommu-qcom-emulate-IDENTITY-domain-for-display-durin.patch.txt](https://github.com/msm8953-mainline/linux/files/11789374/0009-iommu-qcom-emulate-IDENTITY-domain-for-display-durin.patch.txt)
[0010-iommu-qcom-restore-IOMMU-state-if-needed.patch.txt](https://github.com/msm8953-mainline/linux/files/11789380/0010-iommu-qcom-restore-IOMMU-state-if-needed.patch.txt)
[0011-iommu-qcom-defer-probe-until-scm-is-available.patch.txt](https://github.com/msm8953-mainline/linux/files/11789381/0011-iommu-qcom-defer-probe-until-scm-is-available.patch.txt)
[0012-MSM8953-drm-panel-Generate-using-linux-mdss-dsi-pane.patch.txt](https://github.com/msm8953-mainline/linux/files/11789382/0012-MSM8953-drm-panel-Generate-using-linux-mdss-dsi-pane.patch.txt)
[0013-squash-MSM8953-drm-panel-Generate-using-linux-mdss-d.patch.txt](https://github.com/msm8953-mainline/linux/files/11789386/0013-squash-MSM8953-drm-panel-Generate-using-linux-mdss-d.patch.txt)
[0014-drm-panel-fairphone-fp3-hx83112b-Swab-brightness-val.patch.txt](https://github.com/msm8953-mainline/linux/files/11789388/0014-drm-panel-fairphone-fp3-hx83112b-Swab-brightness-val.patch.txt)
[0015-squash-MSM8953-drm-panel-Generate-using-linux-mdss-d.patch.txt](https://github.com/msm8953-mainline/linux/files/11789391/0015-squash-MSM8953-drm-panel-Generate-using-linux-mdss-d.patch.txt)
[0016-arm64-dts-qcom-msm8953-add-IOMMUs.patch.txt](https://github.com/msm8953-mainline/linux/files/11789392/0016-arm64-dts-qcom-msm8953-add-IOMMUs.patch.txt)
[0017-arm64-dts-qcom-msm8953-add-GPU.patch.txt](https://github.com/msm8953-mainline/linux/files/11789393/0017-arm64-dts-qcom-msm8953-add-GPU.patch.txt)
[0018-arm64-dts-qcom-sdm632-fairphone-fp3-add-display.patch.txt](https://github.com/msm8953-mainline/linux/files/11789395/0018-arm64-dts-qcom-sdm632-fairphone-fp3-add-display.patch.txt)
[0019-fixup-arm64-dts-qcom-sdm632-fairphone-fp3-add-displa.patch.txt](https://github.com/msm8953-mainline/linux/files/11789396/0019-fixup-arm64-dts-qcom-sdm632-fairphone-fp3-add-displa.patch.txt)
[0020-interconnect-icc-rpm-Add-option-to-get-regmap-from-p.patch.txt](https://github.com/msm8953-mainline/linux/files/11789398/0020-interconnect-icc-rpm-Add-option-to-get-regmap-from-p.patch.txt)
[0021-dt-bindings-interconnect-Add-bindings-for-Qualcomm-M.patch.txt](https://github.com/msm8953-mainline/linux/files/11789399/0021-dt-bindings-interconnect-Add-bindings-for-Qualcomm-M.patch.txt)
[0022-arm64-dts-qcom-msm8953-add-interconnect.patch.txt](https://github.com/msm8953-mainline/linux/files/11789401/0022-arm64-dts-qcom-msm8953-add-interconnect.patch.txt)
[0023-interconnect-qcom-Add-MSM8953-interconnect-provider-.patch.txt](https://github.com/msm8953-mainline/linux/files/11789403/0023-interconnect-qcom-Add-MSM8953-interconnect-provider-.patch.txt)
[0024-Revert-drm-msm-dsi-Stop-unconditionally-powering-up-.patch.txt](https://github.com/msm8953-mainline/linux/files/11789405/0024-Revert-drm-msm-dsi-Stop-unconditionally-powering-up-.patch.txt)
[0025-arm64-dts-qcom-msm8953-add-GPU-operating-points.patch.txt](https://github.com/msm8953-mainline/linux/files/11789408/0025-arm64-dts-qcom-msm8953-add-GPU-operating-points.patch.txt)
[0026-clk-qcom-gcc-msm8953-don-t-cache-rate-for-dsi-clocks.patch.txt](https://github.com/msm8953-mainline/linux/files/11789409/0026-clk-qcom-gcc-msm8953-don-t-cache-rate-for-dsi-clocks.patch.txt)
[0027-clk-qcom-rcg2-round-parent-rate-when-SET_RATE_PARENT.patch.txt](https://github.com/msm8953-mainline/linux/files/11789411/0027-clk-qcom-rcg2-round-parent-rate-when-SET_RATE_PARENT.patch.txt)

A warning from drm_fb_helper_damage work does not occur here.

z3ntu, 2023-08-23:
FYI some of the iommu patches appear to have been merged https://lore.kernel.org/linux-arm-msm/169158176438.300170.11135849217509321045.b4-ty@kernel.org/

Hopefully will be part of v6.5 kernel.

z3ntu, 2023-10-12:
Worked on this recently, preparing some commits for upstream... Also posted the last missing IOMMU patch yesterday: https://lore.kernel.org/linux-arm-msm/20231011-msm8953-iommu-restore-v1-1-48a0c93809a2@z3ntu.xyz/T/

If everything goes well I should be able to send the GPU patches upstream soon :)

barni2000, 2024-05-11:
@z3ntu Is there anything left? If i know correctly it is in 6.9.

closed 2024-05-11 by barni2000