msm8953-mainline/linux #62: Upstream device dts files

z3ntu, 2022-11-16:
Initial work in #58

z3ntu, 2022-11-16:
Dropping for now sdm450-samsung-a6plte-r4.dts because of too many open issues. Commit for reference: https://github.com/msm8953-mainline/linux/pull/58/commits/92d66a8480188904ba4623c57b8383fdb5dbb796

z3ntu, 2022-11-17:
Seems most dts files we've added are specified to be licensed under GPL-2.0-only license, while the base `msm8953.dtsi` is licensed under BSD-3-Clause which doesn't make much sense; and generally upstream Linux wants dts(i) files to be BSD-licensed.

I'd also remove the copyright lines for Linux Foundation from dts as (to my knowledge) noone here works for Linux Foundation so this line is kind of bogus. I've heard of some that they don't want their name in the copyright line, so unless you tell me I'm not going to add your name there instead and just remove it.

So please confirm the license change (and additionally if you want your name in copyright line):

- [ ] @julianbraha (sdm450-motorola-ali)
- [ ] @SirSireesh (msm8953-motorola-potter)
- [ ] @alikates (msm8953-xiaomi-daisy)
- [ ] @JIaxyga (msm8953-xiaomi-tissot)
- [ ] @jenneron (msm8953-xiaomi-tissot)
- [ ] @fekz115 (msm8953-xiaomi-vince)
- [ ] @M0Rf30 (msm8953-xiaomi-vince)
- [ ] @Ultracoolguy (sdm632-motorola-ocean)

Ultracoolguy, 2022-11-17:
Personally I fine with my name. You can add Gabriela David.

fekz115, 2022-11-17:
I'm fine with licensing
Eugene Lepshy fekz115@gmail.com for vince

julianbraha, 2022-11-17:
Fine by me,
Julian Braha julianbraha@gmail.com for motorola-ali

M0Rf30, 2022-11-17:
Fine with SPDX licensing
Gianluca Boiano morf3089@gmail.com for vince

alikates, 2022-11-17:
It's fine for me with daisy,
Alejandro Tafalla

SirSireesh, 2022-11-17:
I'm fine with BSD for potter, Sireesh Kodali

JIaxyga, 2022-11-17:
[image]
It's fine for jenneron too

z3ntu, 2022-11-17:
@julianbraha Can we cut down `qcom,board-id` for motorola-ali down to one item only?
```
+       qcom,board-id = <0x41 0xb1a0>,
+                       <0x42 0xb1a0>,
+                       <0x43 0xc100>,
+                       <0x42 0xb1b0>,
+                       <0x43 0xc200>,
+                       <0x44 0xc200>,
+                       <0x42 0xb200>,
+                       <0x43 0xb200>;
```

julianbraha, 2022-11-19:
> Can we cut down qcom,board-id for motorola-ali down to one item only?

I think there's multiple qcom,board-id values because Motorola released the phone in multiple regions with only slight modifications. Then again, I don't have all of the variants to test with, so should I just figure out which of the values corresponds to the model I have?

z3ntu, 2022-11-19:
@julianbraha Yeah, if you give me e.g. `qcom,board-id = <0x43 0xb200>;` and that works on your phone, I'm happy. Arguing upstream for allowing multiple board ids for this phone might be a bit tedious since I personally don't have too many feelings for that device :wink:

julianbraha, 2022-11-19:
@z3ntu according to the [downstream kernel device tree for my specific variant (PVT2)](https://github.com/MotorolaMobilityLLC/kernel-msm/blob/e07af7cbbbfbaf37f66ead304611f97788bb2d0e/arch/arm/boot/dts/qcom/sdm450-ali-pvt2.dts), the `qcom,board-id` value is `<0x43 0xC200>, <0x44 0xC200>`. But after some testing, only `<0x43 0xc200>` actually boots. 

So for a single value:
`qcom,board-id = <0x43 0xc200>;`

z3ntu, 2022-11-19:
Thanks everyone, v2 has been sent! https://lore.kernel.org/linux-arm-msm/20221119203758.888207-1-luca@z3ntu.xyz/T/

z3ntu, 2022-12-07:
v3 with some minor changes: https://lore.kernel.org/linux-arm-msm/20221207-msm8953-6-1-next-dtbs-v3-v3-0-a64b3b0af0eb@z3ntu.xyz/T/
Hopefully the last one :) I don't think it'll make the v5.2 merge window though since the second qcom PR has already been sent, so 5.3 hopefully :crossed_fingers:

z3ntu, 2023-04-27:
Right :)

closed 2023-04-27 by z3ntu