msm8953-mainline/linux #68: panel-fairphone-fp3-hx83112b.c: errors at compile time

M0Rf30, 2022-12-23:
```
drivers/gpu/drm/panel/msm8953-generated/panel-fairphone-fp3-hx83112b.c:435:19: error: initialization of 'void (*)(struct mipi_dsi_device *)' from incompatible pointer type 'int (*)(struct mipi_dsi_device *)' [-Werror=incompatible-pointer-types]
  435 |         .remove = djn_hx83112b_remove,
      |                   ^~~~~~~~~~~~~~~~~~~
drivers/gpu/drm/panel/msm8953-generated/panel-fairphone-fp3-hx83112b.c:435:19: note: (near initialization for 'djn_hx83112b_driver.remove')
cc1: some warnings being treated as errors
```

alikates, 2022-12-23:
I probably forgot to add the fp3 dts to the panel generator

z3ntu, 2022-12-23:
Update lmpdg submodule in the panel drivers repo so remove callback returns void instead of int

alikates, 2022-12-23:
Yeah, i updated it in my local tree but havent pushed it yet

M0Rf30, 2022-12-26:
Solved with latest commits on 6.1.0 branch. thanks