msm8953-mainline/linux #69: hynix SR556 camera sensor: errors at compile time

M0Rf30, 2022-12-23:
this should be the offending commit https://github.com/msm8953-mainline/linux/commit/25f6279086277e9087cbcc8f813186258e1f55f9 
```
drivers/media/i2c/sr556.c:1042:19: error: initialization of 'void (*)(struct i2c_client *)' from incompatible pointer type 'int (*)(struct i2c_client *)' [-Werror=incompatible-pointer-types]
 1042 |         .remove = sr556_remove,
      |                   ^~~~~~~~~~~~
drivers/media/i2c/sr556.c:1042:19: note: (near initialization for 'sr556_i2c_driver.remove')
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:250: drivers/media/i2c/sr556.o] Error 1
```
@vidly

z3ntu, 2022-12-23:
Update remove method to return void instead of int.

M0Rf30, 2022-12-26:
Solved with latest commits on 6.1.0 branch. thanks

closed 2022-12-26 by M0Rf30