msm8953-mainline/linux #70: s5k2xx camera sensor driver: errors at compile time (6.1.0)

M0Rf30, 2022-12-23:
offending commit https://github.com/msm8953-mainline/linux/commit/257fb45a206b862b781bba44063739773cd50925

```
drivers/media/i2c/s5k2xx.c:1464:19: error: initialization of 'void (*)(struct i2c_client *)' from incompatible pointer type 'int (*)(struct i2c_client *)' [-Werror=incompatible-pointer-types]
 1464 |         .remove = s5k2xx_remove,
      |                   ^~~~~~~~~~~~~
drivers/media/i2c/s5k2xx.c:1464:19: note: (near initialization for 's5k2xx_i2c_driver.remove')
cc1: some warnings being treated as errors
```
@Vidly @alikates

z3ntu, 2022-12-23:
Update remove method to return void instead of int.

M0Rf30, 2022-12-26:
Solved with latest commits on 6.1.0 branch. thanks

closed 2022-12-26 by M0Rf30