msm8953-mainline/linux #72: v6.1.0 testing

alikates, 2023-01-07:
Hi, can we get the kernel tested on all devices so we can update the aport package?

Tested on:
- [ ] mido
- [x] daisy
- [ ] tissot
- [ ] ocean
- [ ] vince
- [ ] fp3
- [ ] a6plte

(tell me if i'm missing any)

M0Rf30, 2023-01-07:
Done. need to add this https://github.com/msm8953-mainline/linux/pull/71
it will fixes errors encountered on regions overlap.

alikates, 2023-01-07:
I just merged your PR. I also cherry-picked the fixes for the missing sound cards.

M0Rf30, 2023-01-07:
Thanks. I updated also pmaports MR
Could be useful to share built apk that will be available here (at the end of CI process)
https://gitlab.com/M0Rf30/pmaports/-/jobs/3568527126

z3ntu, 2023-02-06:
I believe we can close this now since 6.1 has been merged in pmaports?

closed 2023-02-10 by alikates