msm8953-mainline/linux #74: How to show the kernel log when the kernel booted

bzy-080408, 2023-01-21:
I have a huawei msm8953 device and I tried to port the mainline kernel on it.I built it and booted it,but I weit about 10 second and it reboot into a QUALCOMM mode,and it is not the 9008/9006 mode.I don`t know that what mode it is : (    .The UART has been disabled in bootloader and it cannot enable,I don`t know that where I can find the log.......

M0Rf30, 2023-01-21:
9008/9006 are QUALCOMM mode for [Sahara protocol](https://en.wikipedia.org/wiki/Qualcomm_EDL_mode) (low-level interaction with the device). You need to use some tools that support it. You can find the proprietary, official tools around the web or also an open-source project like [this](https://github.com/bkerler/Loaders)

bzy-080408, 2023-01-22:
is looks like not 9008 or 9006 model and the diver on Windows 10/11 cannot auto install

Clrs17, 2023-04-18:
Is it a Huawei Milan (MLA-XXX) device? Have you successfully booted the device?

bzy-080408, 2023-04-22:
> Is it a Huawei Milan (MLA-XXX) device? Have you successfully booted the device?

what is huawei milan device？ Can you show me the specific model？ e.g. MLA-AL10

Clrs17, 2023-04-22:
> > Is it a Huawei Milan (MLA-XXX) device? Have you successfully booted the device?
> 
> what is huawei milan device？ Can you show me the specific model？ e.g. MLA-AL10

mla-al10

bzy-080408, 2023-04-22:
> > > Is it a Huawei Milan (MLA-XXX) device? Have you successfully booted the device?
> > 
> > what is huawei milan device？ Can you show me the specific model？ e.g. MLA-AL10
> 
> mla-al10

I have the same device and tryed to boot kernel,but I booted unsuccessfully. I think the device-tree has some bug.

bzy-080408, 2023-04-22:
Btw，你能用中文吗？ 要不加个微信聊？

M0Rf30, 2023-04-27:
This can be closed, there's nothing related to the title

closed 2023-06-08 by bzy-080408