msm8953-mainline/linux #79: error on 6.3 compilation due to sm5708 (used by sdm450-samsung-a6plte-r4.dts)

M0Rf30, 2023-03-25:
```
../drivers/power/supply/sm5708_fuelgauge.c:769:23: error: initialization of 'int (*)(struct i2c_client *)' from incompatible pointer type 'int (*)(struct i2c_client *, const struct i2c_device_id *)' [-Werror=incompatible-pointer-types]
  769 |         .probe      = sm5708_battery_probe,
      |                       ^~~~~~~~~~~~~~~~~~~~
../drivers/power/supply/sm5708_fuelgauge.c:769:23: note: (near initialization for 'sm5708_fuelgauge_driver.<anonymous>.probe')
```

M0Rf30, 2023-03-25:
I'm using latest bbf8d93f51b50cfb1987f92609d44ecaaea283cc commit

M0Rf30, 2023-03-25:
I also add this
```
  CC [M]  sound/soc/qcom/qdsp6/q6voice.o
../sound/soc/qcom/qdsp6/q6cvp.c:141:19: error: initialization of 'void (*)(struct apr_device *)' from incompatible pointer type 'int (*)(struct apr_device *)' [-Werror=incompatible-pointer-types]
  141 |         .remove = q6voice_common_remove,
      |                   ^~~~~~~~~~~~~~~~~~~~~
../sound/soc/qcom/qdsp6/q6cvp.c:141:19: note: (near initialization for 'qcom_q6cvp_driver.remove')
cc1: some warnings being treated as errors
make[6]: *** [../scripts/Makefile.build:252: sound/soc/qcom/qdsp6/q6cvp.o] Error 1
make[6]: *** Waiting for unfinished jobs....
  AR      sound/soc/sunxi/built-in.a
../sound/soc/qcom/qdsp6/q6cvs.c:24:19: error: initialization of 'void (*)(struct apr_device *)' from incompatible pointer type 'int (*)(struct apr_device *)' [-Werror=incompatible-pointer-types]
   24 |         .remove = q6voice_common_remove,
      |                   ^~~~~~~~~~~~~~~~~~~~~
../sound/soc/qcom/qdsp6/q6cvs.c:24:19: note: (near initialization for 'qcom_q6cvs_driver.remove')
  AR      sound/soc/tegra/built-in.a
cc1: some warnings being treated as errors
  AR      sound/soc/ti/built-in.a
make[6]: *** [../scripts/Makefile.build:252: sound/soc/qcom/qdsp6/q6cvs.o] Error 1
  AR      sound/soc/uniphier/built-in.a
  AR      sound/soc/ux500/built-in.a
  AR      sound/soc/xilinx/built-in.a
  AR      sound/soc/xtensa/built-in.a
../sound/soc/qcom/qdsp6/q6mvm.c:162:19: error: initialization of 'void (*)(struct apr_device *)' from incompatible pointer type 'int (*)(struct apr_device *)' [-Werror=incompatible-pointer-types]
  162 |         .remove = q6mvm_remove,
      |                   ^~~~~~~~~~~~
../sound/soc/qcom/qdsp6/q6mvm.c:162:19: note: (near initialization for 'qcom_q6mvm_driver.remove')
cc1: some warnings being treated as errors
make[6]: *** [../scripts/Makefile.build:252: sound/soc/qcom/qdsp6/q6mvm.o] Error 1
make[5]: *** [../scripts/Makefile.build:494: sound/soc/qcom/qdsp6] Error 2
make[4]: *** [../scripts/Makefile.build:494: sound/soc/qcom] Error 2
make[3]: *** [../scripts/Makefile.build:494: sound/soc] Error 2
make[2]: *** [../scripts/Makefile.build:494: sound] Error 2
make[2]: *** Waiting for unfinished jobs....
```

M0Rf30, 2023-03-25:
and this 
```
../drivers/extcon/extcon-sm5502.c:265:10: error: 'const struct regmap_irq_chip' has no member named 'mask_invert'; did you mean 'ack_invert'?
  265 |         .mask_invert            = false,
      |          ^~~~~~~~~~~
      |          ack_invert
make[4]: *** [../scripts/Makefile.build:252: drivers/extcon/extcon-sm5502.o] Error 1
make[3]: *** [../scripts/Makefile.build:494: drivers/extcon] Error 2
make[3]: *** Waiting for unfinished jobs....
```

z3ntu, 2023-03-25:
```
../drivers/power/supply/sm5708_fuelgauge.c:769:23: error: initialization of 'int (*)(struct i2c_client *)' from incompatible pointer type 'int (*)(struct i2c_client *, const struct i2c_device_id *)' [-Werror=incompatible-pointer-types]
  769 |         .probe      = sm5708_battery_probe,
```
Remove second parameter from probe function

---

```
../sound/soc/qcom/qdsp6/q6cvp.c:141:19: error: initialization of 'void (*)(struct apr_device *)' from incompatible pointer type 'int (*)(struct apr_device *)' [-Werror=incompatible-pointer-types]
  141 |         .remove = q6voice_common_remove,
```
Change return type to void, change `return 0;` to `return;` and things like that

---

```
../drivers/extcon/extcon-sm5502.c:265:10: error: 'const struct regmap_irq_chip' has no member named 'mask_invert'; did you mean 'ack_invert'?
  265 |         .mask_invert            = false,
```

Just drop the line, see https://lore.kernel.org/linux-arm-msm/20220620200644.1961936-34-aidanmacdonald.0x0@gmail.com/

alikates, 2023-03-27:
I just updated it, please force-pull

M0Rf30, 2023-03-27:
This is still present (commit f8a101dab3c6aa8cc6faac5c5aaaacbf7b3b221d)
```
sound/soc/qcom/qdsp6/q6mvm.c: In function 'q6mvm_remove':
sound/soc/qcom/qdsp6/q6mvm.c:151:16: error: void value not ignored as it ought to be
  151 |         return q6voice_common_remove(adev);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
sound/soc/qcom/qdsp6/q6mvm.c: At top level:
sound/soc/qcom/qdsp6/q6mvm.c:162:19: error: initialization of 'void (*)(struct apr_device *)' from incompatible pointer type 'int (*)(struct apr_device *)' [-Werror=incompatible-pointer-types]
  162 |         .remove = q6mvm_remove,
      |                   ^~~~~~~~~~~~
sound/soc/qcom/qdsp6/q6mvm.c:162:19: note: (near initialization for 'qcom_q6mvm_driver.remove')
sound/soc/qcom/qdsp6/q6mvm.c: In function 'q6mvm_remove':
sound/soc/qcom/qdsp6/q6mvm.c:152:1: error: control reaches end of non-void function [-Werror=return-type]
  152 | }
      | ^
      
```

M0Rf30, 2023-03-27:
commit 25dbeffa4de5c977446988b4b469bd7358a7c86a compiles succesfully

closed 2023-03-27 by M0Rf30