msm8953-mainline/linux #82: please add redmi 6pro support

Mr-Yuling, 2023-04-02:
it is same to mi a2 lite(daisy) ,i use daisy config but can not display

alikates, 2023-04-02:
Are you using lk2nd bootloader?

Mr-Yuling, 2023-04-02:
> Are you using lk2nd bootloader?

yes,I use daisy lk2nd,but it is shows redmi note 4x

fekz115, 2023-04-02:
Thanks for your report. As we don't have your device, we can only help you to add support by yourself, but you have sakura iirc, so I think we can just create dts for lk2nd and reuse daisy's kernel dts. Please, send us photo with information from lk2dn and your msm-id and board-id(if you don't know how to get it, you can just upload your fdt from downstream from path /sys/firmware/fdt)

Mr-Yuling, 2023-04-02:
> Thanks for your report. As we don't have your device, we can only help you to add support by yourself, but you have sakura iirc, so I think we can just create dts for lk2nd and reuse daisy's kernel [...]

I upload on my google drive,please check it
board-id:https://drive.google.com/file/d/1oFNjT_HjcBwJ5nQhOZ0Va5jOqZnQO64i/view?usp=share_link
msm-id:https://drive.google.com/file/d/1vr8LZZ3lnBmdfCnPw71XbRjlJDUy3d0V/view?usp=share_link
lk2nd photo:https://drive.google.com/file/d/1UCU_Ij4Fc1Iq5RGaHJb-U6Xu7pXDb38G/view?usp=share_link

fekz115, 2023-04-02:
So, simply try edit this strings:
https://github.com/msm8953-mainline/lk2nd/blob/main/dts/msm8953-xiaomi-daisy.dts#L10
https://github.com/msm8953-mainline/linux/blob/msm8953-6.1.0/main/arch/arm64/boot/dts/qcom/msm8953-xiaomi-daisy.dts#L26
to this:
`qcom,board-id= <0x1000b 0x09>, <0x1000b 0>;`
If it helps then create PRs

closed 2023-04-04 by Mr-Yuling