msm8953-mainline/linux #83: xiaomi-vince   Shutdown restart stuck.

wu17481748, 2023-04-04:
[image]

z3ntu, 2023-04-12:
Not sure what kernel version you're on but on 6.1 branch (whatever is in pmaports) on reboot this crash happens:
```
[  275.609913] Internal error: Oops: 000000009600004f [#1] PREEMPT SMP
[  275.609939] Modules linked in: wcn36xx mac80211 libarc4 btqcomsmd btqca bluetooth cfg80211 ecdh_generic ecc uinput nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nf_conntrack nf_defrag_ipv4 nf_defrag_ipv6 wcnss_ctrl ipv6 q6asm_dai q6routing q6voice_dai q6afe_clocks q6afe_dai q6voice q6adm q6asm q6afe q6mvm q6cvs q6cvp snd_q6dsp_common q6core q6voice_common apr pdr_interface qrtr_smd venus_dec venus_enc videobuf2_dma_contig videobuf2_memops nxp_nci_i2c nxp_nci nci nfc crct10dif_ce snd_soc_msm8916_digital qcom_rng qcom_q6v5_pas rfkill qcom_q6v5 qrtr venus_core v4l2_mem2mem qcom_wcnss_pil qcom_sysmon videobuf2_v4l2 qmi_helpers qcom_spmi_vadc videobuf2_common qcom_common qcom_vadc_common rtc_pm8xxx qcom_pon qcom_pil_info qcom_spmi_temp_alarm videodev reboot_mode mc socinfo rmtfs_mem himax_hx83112b panel_fairphone_fp3_hx83112b msm gpu_sched mdt_loader drm_display_helper drm_kms_helper drm drm_panel_orientation_quirks cfbcopyarea fb_sys_fops sysimgblt sysfillrect cfbimgblt
[  275.610158]  syscopyarea cfbfillrect
[  275.702130] CPU: 1 PID: 2704 Comm: init Tainted: G        W          6.1.0-postmarketos-qcom-msm8953 #1-postmarketos-qcom-msm8953
[  275.705874] Hardware name: Fairphone 3 (DT)
[  275.717406] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  275.721402] pc : mutex_lock+0x38/0x64
[  275.728340] lr : drm_atomic_helper_shutdown+0x110/0x124 [drm_kms_helper]
[  275.732166] sp : ffff80000c7bbbb0
[  275.738931] x29: ffff80000c7bbbb0 x28: ffff000017907000 x27: 0000000000000000
[  275.742151] x26: 0000000000000000 x25: 0000000000000001 x24: ffff80000910b030
[  275.749267] x23: ffff000011f07490 x22: ffff800009165278 x21: ffff0000117aa410
[  275.756385] x20: ffff000011f07410 x19: ffff800000c65bb8 x18: ffff000012b8e50c
[  275.763503] x17: 000000040044ffff x16: 00500072b5503510 x15: 0000000000000000
[  275.770621] x14: ffff0000114d3800 x13: ffff8000f0d05000 x12: 0000000034d4d91d
[  275.777740] x11: 0000000000000040 x10: ffff800008fede40 x9 : ffff800008fede38
[  275.784857] x8 : ffff000011800028 x7 : 0000000000000000 x6 : 0000000000000000
[  275.791977] x5 : ffff000012349468 x4 : 0000000000000000 x3 : ffff800000c65d20
[  275.799094] x2 : ffff000017907000 x1 : 0000000000000000 x0 : 0000000000000000
[  275.806212] Call trace:
[  275.813323]  mutex_lock+0x38/0x64
[  275.815580]  msm_drv_shutdown+0x2c/0x40 [msm]
[  275.819053]  platform_shutdown+0x24/0x3c
[  275.823394]  device_shutdown+0x148/0x260
[  275.827386]  kernel_restart+0x40/0xac
[  275.831292]  __do_sys_reboot+0x1e0/0x264
[  275.834851]  __arm64_sys_reboot+0x24/0x30
[  275.838844]  invoke_syscall+0x48/0x114
[  275.842751]  el0_svc_common.constprop.0+0x44/0xec
[  275.846398]  do_el0_svc+0x2c/0xd0
[  275.851169]  el0_svc+0x2c/0x84
[  275.854469]  el0t_64_sync_handler+0xf4/0x120
[  275.857421]  el0t_64_sync+0x18c/0x190
[  275.861852] Code: f9800071 c85ffc60 ca010004 b5000064 (c8047c62) 
[  275.865414] ---[ end trace 0000000000000000 ]---
[  276.872413] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000000
[  277.072991] Kernel Offset: disabled
[  277.073006] CPU features: 0x00000,00814080,0000421b
[  277.075269] Memory Limit: none
[  277.080131] Rebooting in 120 seconds..
```

M0Rf30, 2023-04-12:
I will take a look on my vince in the next days

alikates, 2023-04-13:
This also happens on daisy with 6.1, but looks like its fixed in 6.3

wu17481748, 2023-04-13:
> 不知道你在哪个内核版本上，但是在6.1分支上（不管是pmaports中的什么）重新启动时会发生崩溃：
> 
> ```
> [  275.609913] Internal error: Oops: 000000009600004f [#1] PREEMPT SMP
> [  275.609939] Modules linked in: wcn36xx mac80211 libarc4 btqcomsmd btqca bluetooth cfg80211 ecdh_generic ecc uinput nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nf_conntrack nf_defrag_ipv4 nf_defrag_ipv6 wcnss_ctrl ipv6 q6asm_dai q6routing q6voice_dai q6afe_clocks q6afe_dai q6voice q6adm q6asm q6afe q6mvm q6cvs q6cvp snd_q6dsp_common q6core q6voice_common apr pdr_interface qrtr_smd venus_dec venus_enc videobuf2_dma_contig videobuf2_memops nxp_nci_i2c nxp_nci nci nfc crct10dif_ce snd_soc_msm8916_digital qcom_rng qcom_q6v5_pas rfkill qcom_q6v5 qrtr venus_core v4l2_mem2mem qcom_wcnss_pil qcom_sysmon videobuf2_v4l2 qmi_helpers qcom_spmi_vadc videobuf2_common qcom_common qcom_vadc_common rtc_pm8xxx qcom_pon qcom_pil_info qcom_spmi_temp_alarm videodev reboot_mode mc socinfo rmtfs_mem himax_hx83112b panel_fairphone_fp3_hx83112b msm gpu_sched mdt_loader drm_display_helper drm_kms_helper drm drm_panel_orientation_quirks cfbcopyarea fb_sys_fops sysimgblt sysfillrect cfbimgblt
> [  275.610158]  syscopyarea cfbfillrect
> [  275.702130] CPU: 1 PID: 2704 Comm: init Tainted: G        W          6.1.0-postmarketos-qcom-msm8953 #1-postmarketos-qcom-msm8953
> [  275.705874] Hardware name: Fairphone 3 (DT)
> [  275.717406] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
> [  275.721402] pc : mutex_lock+0x38/0x64
> [  275.728340] lr : drm_atomic_helper_shutdown+0x110/0x124 [drm_kms_helper]
> [  275.732166] sp : ffff80000c7bbbb0
> [  275.738931] x29: ffff80000c7bbbb0 x28: ffff000017907000 x27: 0000000000000000
> [  275.742151] x26: 0000000000000000 x25: 0000000000000001 x24: ffff80000910b030
> [  275.749267] x23: ffff000011f07490 x22: ffff800009165278 x21: ffff0000117aa410
> [  275.756385] x20: ffff000011f07410 x19: ffff800000c65bb8 x18: ffff000012b8e50c
> [  275.763503] x17: 000000040044ffff x16: 00500072b5503510 x15: 0000000000000000
> [  275.770621] x14: ffff0000114d3800 x13: ffff8000f0d05000 x12: 0000000034d4d91d
> [  275.777740] x11: 0000000000000040 x10: ffff800008fede40 x9 : ffff800008fede38
> [  275.784857] x8 : ffff000011800028 x7 : 0000000000000000 x6 : 0000000000000000
> [  275.791977] x5 : ffff000012349468 x4 : 0000000000000000 x3 : ffff800000c65d20
> [  275.799094] x2 : ffff000017907000 x1 : 0000000000000000 x0 : 0000000000000000
> [  275.806212] Call trace:
> [  275.813323]  mutex_lock+0x38/0x64
> [  275.815580]  msm_drv_shutdown+0x2c/0x40 [msm]
> [  275.819053]  platform_shutdown+0x24/0x3c
> [  275.823394]  device_shutdown+0x148/0x260
> [  275.827386]  kernel_restart+0x40/0xac
> [  275.831292]  __do_sys_reboot+0x1e0/0x264
> [  275.834851]  __arm64_sys_reboot+0x24/0x30
> [  275.838844]  invoke_syscall+0x48/0x114
> [  275.842751]  el0_svc_common.constprop.0+0x44/0xec
> [  275.846398]  do_el0_svc+0x2c/0xd0
> [  275.851169]  el0_svc+0x2c/0x84
> [  275.854469]  el0t_64_sync_handler+0xf4/0x120
> [  275.857421]  el0t_64_sync+0x18c/0x190
> [  275.861852] Code: f9800071 c85ffc60 ca010004 b5000064 (c8047c62) 
> [  275.865414] ---[ end trace 0000000000000000 ]---
> [  276.872413] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000000
> [  277.072991] Kernel Offset: disabled
> [  277.073006] CPU features: 0x00000,00814080,0000421b
> [  277.075269] Memory Limit: none
> [  277.080131] Rebooting in 120 seconds..
> ```

Solved, using Xiaomi Prime 6.3 kernel and using 6.0 device tree will not cause shutdown, restart and jamming.

M0Rf30, 2023-04-24:
This can be closed

closed 2023-04-24 by z3ntu