msm8953-mainline/linux #100: linux-6.3 needs cache-size in devicetree

spasswolf, 2023-05-13:
Since linux-6.3 booting linux on the fairphone-fp3 gives the following error messages:
```
```
This patch fixes this on the fairphone-fp3, but similar patches will be needed for other devices:
```
commit 46d99d083306e88af47c5943b8229551fd63d720
Author: Bert Karwatzki <spasswolf@web.de>
Date:   Sat May 13 13:04:10 2023 +0200

    Add {i,d}-cache-size properties to the CPU nodes, cache-size
    and unified-cache properties to the L2 nodes.

diff --git a/arch/arm64/boot/dts/qcom/msm8953.dtsi b/arch/arm64/boot/dts/qcom/msm8953.dtsi
index 8328bac9f690..a88df4f14299 100644
--- a/arch/arm64/boot/dts/qcom/msm8953.dtsi
+++ b/arch/arm64/boot/dts/qcom/msm8953.dtsi
@@ -284,11 +284,13 @@ CLUSTER_PC: cluster-power-collapse {
 		L2_0: l2-cache-0 {
 			compatible = "cache";
 			cache-level = <2>;
+			cache-unified;
 		};
 
 		L2_1: l2-cache-1 {
 			compatible = "cache";
 			cache-level = <2>;
+			cache-unified;
 		};
 	};
 
diff --git a/arch/arm64/boot/dts/qcom/sdm632.dtsi b/arch/arm64/boot/dts/qcom/sdm632.dtsi
index 8f0c8e686733..ff9f943bf2ef 100644
--- a/arch/arm64/boot/dts/qcom/sdm632.dtsi
+++ b/arch/arm64/boot/dts/qcom/sdm632.dtsi
@@ -148,29 +148,43 @@ cpu7-thermal {
 /*
  * SDM632 uses Kryo 250 instead of Cortex A53
  * CPU0-3 are efficiency cores, CPU4-7 are performance cores
+ * cache sizes taken from
+ * https://en.wikipedia.org/wiki/Kryo#Kryo_250
  */
 &CPU0 {
 	compatible = "qcom,kryo250";
 	clocks = <&apcc_c0_mux>;
 	clock-names = "cpu";
+	i-cache-size = <0x8000>;
+	d-cache-size = <0x8000>;
 };
 
 &CPU1 {
 	compatible = "qcom,kryo250";
 	clocks = <&apcc_c0_mux>;
 	clock-names = "cpu";
+	i-cache-size = <0x8000>;
+	d-cache-size = <0x8000>;
 };
 
 &CPU2 {
 	compatible = "qcom,kryo250";
 	clocks = <&apcc_c0_mux>;
 	clock-names = "cpu";
+	i-cache-size = <0x8000>;
+	d-cache-size = <0x8000>;
 };
 
 &CPU3 {
 	compatible = "qcom,kryo250";
 	clocks = <&apcc_c0_mux>;
 	clock-names = "cpu";
+	i-cache-size = <0x8000>;
+	d-cache-size = <0x8000>;
+};
+
+&L2_0 {
+	cache-size = <0x80000>;
 };
 
 &apcc_pll {
@@ -225,6 +239,8 @@ &CPU4 {
 	clocks = <&apcc_c1_mux>;
 	clock-names = "cpu";
 	power-domains = <&cpr 1>;
+	i-cache-size = <0x8000>;
+	d-cache-size = <0x8000>;
 };
 
 &CPU5 {
@@ -233,6 +249,8 @@ &CPU5 {
 	clocks = <&apcc_c1_mux>;
 	clock-names = "cpu";
 	power-domains = <&cpr 1>;
+	i-cache-size = <0x8000>;
+	d-cache-size = <0x8000>;
 };
 
 &CPU6 {
@@ -241,6 +259,8 @@ &CPU6 {
 	clocks = <&apcc_c1_mux>;
 	clock-names = "cpu";
 	power-domains = <&cpr 1>;
+	i-cache-size = <0x8000>;
+	d-cache-size = <0x8000>;
 };
 
 &CPU7 {
@@ -249,6 +269,12 @@ &CPU7 {
 	clocks = <&apcc_c1_mux>;
 	clock-names = "cpu";
 	power-domains = <&cpr 1>;
+	i-cache-size = <0x8000>;
+	d-cache-size = <0x8000>;
+};
+
+&L2_1 {
+	cache-size = <0x100000>;
 };
 
 &cpr {
```

ultra-azu, 2023-05-13:
It might be best if you put this in a PR.

z3ntu, 2023-05-13:
For the first part there's a patch on the list already: https://lore.kernel.org/linux-arm-msm/20230416101134.95686-3-krzysztof.kozlowski@linaro.org/#Z31arch:arm64:boot:dts:qcom:msm8953.dtsi

spasswolf, 2023-06-20:
The cache-size properties don't seem to be necessary when using linux-next-20230616.

z3ntu, 2023-09-23:
The linked patch has landed in mainline already, I guess we can close this?

closed 2024-04-03 by vldly