msm8953-mainline/linux #106: xiaomi–vince modemmanager and ofono unable to use modem

wu17481748, 2023-06-11:
The sim card can be recognized.
xiaomi-vince:~# qmicli -d qrtr://0 --uim-get-card-status

[qrtr://0] Successfully got card status Provisioning applications: Primary GW: 

slot '1', application '1' Primary 1X: session doesn't exist Secondary GW: session doesn't exist Secondary 1X: session doesn't exist Slot [1]: Card state: 'present' UPIN state: 'not-initialized' UPIN retries: '0' UPUK retries: '0' Application [1]: Application type: 'usim (2)' Application state: 'ready' Application ID: A0:00:00:00:87:10:02:FF:86:FF:FF:89:FF:FF:FF:FF Personalization state: 'ready' UPIN replaces PIN1: 'no' PIN1 state: 'disabled' PIN1 retries: '1' PUK1 retries: '10' PIN2 state: 'enabled-not-verified' PIN2 retries: '3' PUK2 retries: '10' 

sot [2]: Card state: 'error: no-atr-received (3)' UPIN state: 'not-initialized' UPIN retries: '0' UPUK retries: '0'

xiaomi-vince:~# ModemManager --debug

[5572]: [000519809.909721] ModemManager (version 1.20.6) starting in system bus... 
[5572]: [000519809.909880] [charsets] detecting platform iconv() support... 
[5572]: [000519809.909935] [charsets] UTF-8: iconv conversion to/from charset is supported 
[5572]: [000519809.909970] [charsets] UCS-2BE: iconv conversion to/from charset is supported 
[5572]: [000519809.910007] [charsets] ASCII: iconv conversion to/from charset is supported 
[5572]: [000519809.910044] [charsets] ISO8859-1: iconv conversion to/from charset is supported 
[5572]: [000519809.910088] [charsets] CP437: iconv conversion to/from charset is supported 
[5572]: [000519809.910132] [charsets] CP850: iconv conversion to/from charset is supported 
[5572]: [000519809.910165] [charsets] UTF-16BE: iconv conversion to/from charset is supported 
[5572]: [000519809.912440] Full suspend/resume hooks enabled 
[5572]: [000519809.913205] [sleep-monitor-systemd] singleton created 
[5572]: [000519809.917858] bus acquired, creating manager... 
[5572]: [000519809.923192] [auth-provider] singleton created 
[5572]: [000519809.923406] [filter] created 
[5572]: [000519809.923440] [filter] explicit allowlist: yes 
[5572]: [000519809.923464] [filter] explicit blocklist: yes 
[5572]: [000519809.923491] [filter] plugin allowlist: yes 
[5572]: [000519809.923514] [filter] qrtr devices allowed: yes 
[5572]: [000519809.923540] [filter] virtual devices forbidden: yes 
[5572]: [000519809.923567] [filter] net devices allowed: yes 
[5572]: [000519809.923590] [filter] usbmisc devices allowed: yes 
[5572]: [000519809.923613] [filter] rpmsg devices allowed: yes 
[5572]: [000519809.923638] [filter] wwan devices allowed: yes 
[5572]: [000519809.923663] [filter] tty devices: 
[5572]: [000519809.923688] [filter] platform driver check: yes 
[5572]: [000519809.923715] [filter] driver check: yes 
[5572]: [000519809.923743] [filter] cdc-acm interface check: yes 
[5572]: [000519809.923770] [filter] with net check: yes 
[5572]: [000519809.923796] [filter] default: forbidden 
[5572]: [000519809.923906] [plugin-manager] looking for plugins in '/usr/lib/ModemManager' 
[5572]: [000519809.924575] [plugin-manager] loaded shared 'Sierra' utils from '/usr/lib/ModemManager/libmm-shared-sierra.so' 
[5572]: [000519809.924913] [plugin-manager] loaded shared 'Telit' utils from '/usr/lib/ModemManager/libmm-shared-telit.so' 
[5572]: [000519809.925115] [plugin-manager] loaded shared 'Novatel' utils from '/usr/lib/ModemManager/libmm-shared-novatel.so' 
[5572]: [000519809.925408] [plugin-manager] loaded shared 'Xmm' utils from '/usr/lib/ModemManager/libmm-shared-xmm.so' 
[5572]: [000519809.925647] [plugin-manager] loaded shared 'Foxconn' utils from '/usr/lib/ModemManager/libmm-shared-foxconn.so' 
[5572]: [000519809.925855] [plugin-manager] loaded shared 'Option' utils from '/usr/lib/ModemManager/libmm-shared-option.so' 
[5572]: [000519809.926152] [plugin-manager] loaded shared 'Icera' utils from '/usr/lib/ModemManager/libmm-shared-icera.so' 
[5572]: [000519809.926315] [plugin-manager] loaded shared 'Intel' utils from '/usr/lib/ModemManager/libmm-shared-fibocom.so' 
[5572]: [000519809.926886] [plugin-manager] loaded plugin 'nokia-icera' from '/usr/lib/ModemManager/libmm-plugin-nokia-icera.so' 
[5572]: [000519809.926930] [filter] registered plugin allowlist vendor id: 0421 
[5572]: [000519809.927172] [plugin-manager] loaded plugin 'x22x' from '/usr/lib/ModemManager/libmm-plugin-x22x.so' 
[5572]: [000519809.927209] [filter] registered plugin allowlist tag: ID_MM_X22X_TAGGED 
[5572]: [000519809.927240] [filter] registered plugin allowlist vendor id: 1bbb 
[5572]: [000519809.927267] [filter] registered plugin allowlist vendor id: 0b3c 
[5572]: [000519809.927473] [plugin-manager] loaded plugin 'telit' from '/usr/lib/ModemManager/libmm-plugin-telit.so' 
[5572]: [000519809.927509] [filter] registered plugin allowlist vendor id: 1bc7 
[5572]: [000519809.927539] [filter] registered plugin allowlist subsystem vendor id: 17cb:1c5d 
[5572]: [000519809.927731] [plugin-manager] loaded plugin 'gosuncn' from '/usr/lib/ModemManager/libmm-plugin-gosuncn.so' 
[5572]: [000519809.927767] [filter] registered plugin allowlist vendor id: 305a [5572]: [000519809.928017] [plugin-manager] loaded plugin 'quectel' from '/usr/lib/ModemManager/libmm-plugin-quectel.so' 
[5572]: [000519809.928052] [filter] registered plugin allowlist vendor id: 2c7c 
[5572]: [000519809.928082] [filter] registered plugin allowlist vendor id: 1eac 
[5572]: [000519809.928362] [plugin-manager] loaded plugin 'generic' from '/usr/lib/ModemManager/libmm-plugin-generic.so' 
[5572]: [000519809.928599] [plugin-manager] loaded plugin 'thuraya' from '/usr/lib/ModemManager/libmm-plugin-thuraya.so' 
[5572]: [000519809.928635] [filter] registered plugin allowlist vendor id: 1a26 
[5572]: [000519809.928894] [plugin-manager] loaded plugin 'simtech' from '/usr/lib/ModemManager/libmm-plugin-simtech.so' 
[5572]: [000519809.928937] [filter] registered plugin allowlist vendor id: 1e0e 
[5572]: [000519809.929187] [plugin-manager] loaded plugin 'motorola' from '/usr/lib/ModemManager/libmm-plugin-mtk.so' 
[5572]: [000519809.929225] [filter] registered plugin allowlist tag: ID_MM_MTK_TAGGED 
[5572]: [000519809.929429] [plugin-manager] loaded plugin 'sierra' from '/usr/lib/ModemManager/libmm-plugin-sierra.so' 
[... 109 lines cut ...]

[2238]: [1686455444.258363] [plugin-manager] task 0: no more ports to probe

[2238]: [1686455444.258472] [plugin-manager] task 0: finished in '2.501421' seconds

[2238]: [1686455444.258685] [device qcom-soc] creating modem with plugin 'qcom-soc' and '1' ports

[2238]: [1686455444.258816] [plugin/qcom-soc] (qcom-soc) Qualcomm SoC modem found...

[2238]: [1686455444.269901] [qrtr0/qmi] port monitoring enabled in QMI port

[2238]: [1686455444.270120] [modem0] port 'qrtr/qrtr0' grabbed

[2238]: [1686455444.270323] [modem0] cleaning up port 'qrtr/qrtr0'...

[2238]: [1686455444.270613] [modem0] completely disposed

[2238]: [1686455444.270745] [base-manager] couldn't create modem for device 'qcom-soc': Failed to find a net port in the QMI modem

I tried the ofono branch ./list-modem can see the modem, but ./enable-modem reported an error.Invalid apn setting.

https://github.com/msm8953-mainline/ofono

#####
How can I use the modem?

alikates, 2023-06-11:
ModemManager requires mobile data support from the kernel side, even if you only want to use phone calls.

For mobile data to work we first need to fix support for Qualcomm IPA.

If you want you can try the patches from #105 so MM sees a network port and continues loading. Only phone calls/SMS will work though.

wu17481748, 2023-06-11:
> Modemmanager需要内核端的移动数据支持，即使您只想使用电话呼叫。类型
> 
> 为了移动数据的工作，我们首先需要修复对高通IPA的支持。寻求帮助和提示
> 
> 如果你愿意，你可以试试类型为：pr以筛选提取请求#105所以MM看到一个网络端口并继续加载。只有电话/短信可以用。类型

thank you Try this patch then. I wonder if ipa will continue to be repaired until mobile data can be used. very much looking forward to using mobile data.

wu17481748, 2023-06-11:
> ModemManager requires mobile data support from the kernel side, even if you only want to use phone calls.
>
> For mobile data to work we first need to fix support for Qualcomm IPA.
>
> If you want you can [...]

thank you Try this patch then. I wonder if ipa will continue to be repaired until mobile data can be used. very much looking forward to using mobile data.

wu17481748, 2023-06-11:
> ModemManager requires mobile data support from the kernel side, even if you only want to use phone calls.
>
> For mobile data to work we first need to fix support for Qualcomm IPA.
>
> If you want you can [...]

Hello, I don't know why.
Ipa branch mainline kernel is used.

#ModemManager status

1月 30 09:06:05 mobian ModemManager[673]: <warn>  [modem0] couldn't enable interface: 'Couldn't set operating mode: QMI protocol error (52): 'DeviceNotReady''
1月 30 09:06:05 mobian ModemManager[673]: <info>  [modem0] state changed (enabling -> disabled)
1月 30 09:06:05 mobian ModemManager[673]: <info>  [modem0] simple connect started...
1月 30 09:06:05 mobian ModemManager[673]: <info>  [modem0] simple connect state (3/10): enable
1月 30 09:06:05 mobian ModemManager[673]: <info>  [modem0] state changed (disabled -> enabling)
1月 30 09:06:05 mobian ModemManager[673]: <warn>  [modem0] couldn't enable interface: 'Couldn't set operating mode: QMI protocol error (52): 'DeviceNotReady''
1月 30 09:06:05 mobian ModemManager[673]: <info>  [modem0] state changed (enabling -> disabled)
1月 30 09:06:05 mobian ModemManager[673]: <info>  [modem0] simple connect started...
1月 30 09:06:05 mobian ModemManager[673]: <info>  [modem0] simple connect state (3/10): enable
1月 30 09:06:05 mobian ModemManager[673]: <info>  [modem0] state changed (disabled -> enabling)

#nmlci
The qrtr0 adapter interface failed to connect.
qrtr0：已断开
        "qrtr0"                                                                                                   1 连接可用                                                                                                gsm (qrtr, ipa), 硬件

Can't make a phone call or send a text message.

wu17481748, 2023-06-11:
> ModemManager requires mobile data support from the kernel side, even if you only want to use phone calls.
>
> For mobile data to work we first need to fix support for Qualcomm IPA.
>
> If you want you can [...]

thank
Replace the modem file and you can call and send text messages.

barni2000, 2024-05-11:
ModeManager and ipa2-lite handles this now.

closed 2024-05-11 by barni2000