msm8953-mainline/linux #107: Patches for linux-next ()

spasswolf, 2023-06-20:
This is a collection of the patches for msm8953 support in linux-next (currently for linux-next-20230616, but rebasing them to newer version was without problems so far, I started at linux-next-20230609). It would be nicer if I could just push them to a tree, but unfortunately it seems I cannot fork torvalds/linux, because I already have forked msm8953-mainline/linux (which is itself a fork of msm8916-mainline/linux which is a fork of torvalds linux).
GPU patches:
[0001-iommu-qcom-Use-the-asid-read-from-device-tree-if-spe.patch.txt](https://github.com/msm8953-mainline/linux/files/11806356/0001-iommu-qcom-Use-the-asid-read-from-device-tree-if-spe.patch.txt)
[0002-iommu-qcom-Properly-reset-the-IOMMU-context.patch.txt](https://github.com/msm8953-mainline/linux/files/11806358/0002-iommu-qcom-Properly-reset-the-IOMMU-context.patch.txt)
[0003-iommu-qcom-Index-contexts-by-asid-number-to-allow-as.patch.txt](https://github.com/msm8953-mainline/linux/files/11806360/0003-iommu-qcom-Index-contexts-by-asid-number-to-allow-as.patch.txt)
[0004-iommu-qcom-Add-support-for-QSMMUv2-and-QSMMU-500-sec.patch.txt](https://github.com/msm8953-mainline/linux/files/11806365/0004-iommu-qcom-Add-support-for-QSMMUv2-and-QSMMU-500-sec.patch.txt)
[0005-iommu-qcom-Add-support-for-AArch64-IOMMU-pagetables.patch.txt](https://github.com/msm8953-mainline/linux/files/11806370/0005-iommu-qcom-Add-support-for-AArch64-IOMMU-pagetables.patch.txt)
[0006-fixup-iommu-qcom-Add-support-for-AArch64-IOMMU-paget.patch.txt](https://github.com/msm8953-mainline/linux/files/11806372/0006-fixup-iommu-qcom-Add-support-for-AArch64-IOMMU-paget.patch.txt)
[0007-iommu-qcom-avoid-bouncing-power-domain.patch.txt](https://github.com/msm8953-mainline/linux/files/11806373/0007-iommu-qcom-avoid-bouncing-power-domain.patch.txt)
[0008-iommu-qcom-implement-adreno_priv-interface.patch.txt](https://github.com/msm8953-mainline/linux/files/11806375/0008-iommu-qcom-implement-adreno_priv-interface.patch.txt)
[0009-iommu-qcom-emulate-IDENTITY-domain-for-display-durin.patch.txt](https://github.com/msm8953-mainline/linux/files/11806376/0009-iommu-qcom-emulate-IDENTITY-domain-for-display-durin.patch.txt)
[0010-iommu-qcom-restore-IOMMU-state-if-needed.patch.txt](https://github.com/msm8953-mainline/linux/files/11806378/0010-iommu-qcom-restore-IOMMU-state-if-needed.patch.txt)
[0011-iommu-qcom-defer-probe-until-scm-is-available.patch.txt](https://github.com/msm8953-mainline/linux/files/11806379/0011-iommu-qcom-defer-probe-until-scm-is-available.patch.txt)
[0012-MSM8953-drm-panel-Generate-using-linux-mdss-dsi-pane.patch.txt](https://github.com/msm8953-mainline/linux/files/11806380/0012-MSM8953-drm-panel-Generate-using-linux-mdss-dsi-pane.patch.txt)
[0013-squash-MSM8953-drm-panel-Generate-using-linux-mdss-d.patch.txt](https://github.com/msm8953-mainline/linux/files/11806381/0013-squash-MSM8953-drm-panel-Generate-using-linux-mdss-d.patch.txt)
[0014-drm-panel-fairphone-fp3-hx83112b-Swab-brightness-val.patch.txt](https://github.com/msm8953-mainline/linux/files/11806382/0014-drm-panel-fairphone-fp3-hx83112b-Swab-brightness-val.patch.txt)
[0015-squash-MSM8953-drm-panel-Generate-using-linux-mdss-d.patch.txt](https://github.com/msm8953-mainline/linux/files/11806384/0015-squash-MSM8953-drm-panel-Generate-using-linux-mdss-d.patch.txt)
[0016-arm64-dts-qcom-msm8953-add-IOMMUs.patch.txt](https://github.com/msm8953-mainline/linux/files/11806385/0016-arm64-dts-qcom-msm8953-add-IOMMUs.patch.txt)
[0017-arm64-dts-qcom-msm8953-add-GPU.patch.txt](https://github.com/msm8953-mainline/linux/files/11806387/0017-arm64-dts-qcom-msm8953-add-GPU.patch.txt)
[0018-arm64-dts-qcom-sdm632-fairphone-fp3-add-display.patch.txt](https://github.com/msm8953-mainline/linux/files/11806388/0018-arm64-dts-qcom-sdm632-fairphone-fp3-add-display.patch.txt)
[0019-fixup-arm64-dts-qcom-sdm632-fairphone-fp3-add-displa.patch.txt](https://github.com/msm8953-mainline/linux/files/11806390/0019-fixup-arm64-dts-qcom-sdm632-fairphone-fp3-add-displa.patch.txt)
[0020-interconnect-icc-rpm-Add-option-to-get-regmap-from-p.patch.txt](https://github.com/msm8953-mainline/linux/files/11806391/0020-interconnect-icc-rpm-Add-option-to-get-regmap-from-p.patch.txt)
[0021-dt-bindings-interconnect-Add-bindings-for-Qualcomm-M.patch.txt](https://github.com/msm8953-mainline/linux/files/11806393/0021-dt-bindings-interconnect-Add-bindings-for-Qualcomm-M.patch.txt)
[0022-arm64-dts-qcom-msm8953-add-interconnect.patch.txt](https://github.com/msm8953-mainline/linux/files/11806394/0022-arm64-dts-qcom-msm8953-add-interconnect.patch.txt)
[0023-interconnect-qcom-Add-MSM8953-interconnect-provider-.patch.txt](https://github.com/msm8953-mainline/linux/files/11806396/0023-interconnect-qcom-Add-MSM8953-interconnect-provider-.patch.txt)
[0024-Revert-drm-msm-dsi-Stop-unconditionally-powering-up-.patch.txt](https://github.com/msm8953-mainline/linux/files/11806398/0024-Revert-drm-msm-dsi-Stop-unconditionally-powering-up-.patch.txt)
[0025-arm64-dts-qcom-msm8953-add-GPU-operating-points.patch.txt](https://github.com/msm8953-mainline/linux/files/11806400/0025-arm64-dts-qcom-msm8953-add-GPU-operating-points.patch.txt)
[0026-clk-qcom-gcc-msm8953-don-t-cache-rate-for-dsi-clocks.patch.txt](https://github.com/msm8953-mainline/linux/files/11806404/0026-clk-qcom-gcc-msm8953-don-t-cache-rate-for-dsi-clocks.patch.txt)
[0027-clk-qcom-rcg2-round-parent-rate-when-SET_RATE_PARENT.patch.txt](https://github.com/msm8953-mainline/linux/files/11806410/0027-clk-qcom-rcg2-round-parent-rate-when-SET_RATE_PARENT.patch.txt)
WLAN patches:
 [0028-arm64-dts-qcom-sdm632-fairphone-fp3-enable-wcnss-rem.patch.txt](https://github.com/msm8953-mainline/linux/files/11806418/0028-arm64-dts-qcom-sdm632-fairphone-fp3-enable-wcnss-rem.patch.txt)
[0029-net-wireless-wcn36xx-fix-Bad-TX-complete-indication-.patch.txt](https://github.com/msm8953-mainline/linux/files/11806419/0029-net-wireless-wcn36xx-fix-Bad-TX-complete-indication-.patch.txt)
modem (named mpss in linux-next) patches:
[0030-remoteproc-qcom_q6v5_mss-Support-alternate-msm8953-c.patch.txt](https://github.com/msm8953-mainline/linux/files/11806422/0030-remoteproc-qcom_q6v5_mss-Support-alternate-msm8953-c.patch.txt)
[0031-arm64-dts-qcom-sdm632-fairphone-fp3-enable-mpss.patch.txt](https://github.com/msm8953-mainline/linux/files/11806426/0031-arm64-dts-qcom-sdm632-fairphone-fp3-enable-mpss.patch.txt)
IPA v2 patches:
[0032-dmaengine-Add-support-for-immediate-commands-in-the-.patch.txt](https://github.com/msm8953-mainline/linux/files/11806431/0032-dmaengine-Add-support-for-immediate-commands-in-the-.patch.txt)
[0033-dmaengine-qcom-bam_dma-Add-support-for-immediate-com.patch.txt](https://github.com/msm8953-mainline/linux/files/11806435/0033-dmaengine-qcom-bam_dma-Add-support-for-immediate-com.patch.txt)
[0034-net-ipa-Define-2.-versions.patch.txt](https://github.com/msm8953-mainline/linux/files/11806436/0034-net-ipa-Define-2.-versions.patch.txt)
[0035-net-ipa-Rename-gsi_ee_id-to-dma_ee_id.patch.txt](https://github.com/msm8953-mainline/linux/files/11806438/0035-net-ipa-Rename-gsi_ee_id-to-dma_ee_id.patch.txt)
[0036-TODO-net-ipa-Add-IPA-v2.-register-definitions.patch.txt](https://github.com/msm8953-mainline/linux/files/11806439/0036-TODO-net-ipa-Add-IPA-v2.-register-definitions.patch.txt)
[0037-net-ipa-Move-GSI-specific-functions-to-an-ops-struct.patch.txt](https://github.com/msm8953-mainline/linux/files/11806441/0037-net-ipa-Move-GSI-specific-functions-to-an-ops-struct.patch.txt)
[0038-net-ipa-Move-gsi_channel_tre_max-definition-to-gsi_p.patch.txt](https://github.com/msm8953-mainline/linux/files/11806442/0038-net-ipa-Move-gsi_channel_tre_max-definition-to-gsi_p.patch.txt)
[0039-net-ipa-gsi.h-Rename-gsi-to-ipa_dma.patch.txt](https://github.com/msm8953-mainline/linux/files/11806443/0039-net-ipa-gsi.h-Rename-gsi-to-ipa_dma.patch.txt)
[0040-net-ipa-gsi_private.h-Rename-gsi-to-ipa_dma.patch.txt](https://github.com/msm8953-mainline/linux/files/11806446/0040-net-ipa-gsi_private.h-Rename-gsi-to-ipa_dma.patch.txt)
[0041-net-ipa-gsi_trans-Rename-gsi-to-ipa_dma.patch.txt](https://github.com/msm8953-mainline/linux/files/11806449/0041-net-ipa-gsi_trans-Rename-gsi-to-ipa_dma.patch.txt)
[0042-net-ipa-Add-timeout-for-ipa_cmd_pipeline_clear_wait.patch.txt](https://github.com/msm8953-mainline/linux/files/11806450/0042-net-ipa-Add-timeout-for-ipa_cmd_pipeline_clear_wait.patch.txt)
[... cut ...]
[0063-net-ipa-Adjust-the-validity-check-of-ipa-registers-t.patch.txt](https://github.com/msm8953-mainline/linux/files/11806763/0063-net-ipa-Adjust-the-validity-check-of-ipa-registers-t.patch.txt)
[0064-net-ipa-Check-interrupts-for-availability.patch.txt](https://github.com/msm8953-mainline/linux/files/11806764/0064-net-ipa-Check-interrupts-for-availability.patch.txt)
[0065-net-ipa-Add-support-for-IPA-v2.x-interrupts.patch.txt](https://github.com/msm8953-mainline/linux/files/11806765/0065-net-ipa-Add-support-for-IPA-v2.x-interrupts.patch.txt)

Still missing are the cpufreq patches which are already in msm8953-mainline/linux.

Tooniis, 2023-06-20:
> it seems I cannot fork torvalds/linux, because I already have forked msm8953-mainline/linux (which is itself a fork of msm8916-mainline/linux which is a fork of torvalds linux).
```
git remote add torvalds https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git
git fetch torvalds master
git checkout torvalds/master
```

spasswolf, 2023-06-20:
Thanks, I created the patches above by remote adding my github fork to my local clone of linux-next so it's strange that I didn't think of this.

wu17481748, 2023-06-22:
> 成交。在我的github存储库中https://github.com/spasswolf/linux.git现在有一个名为msm8953_iommu_gpu_wlan_modem_ipa的分支，它包含上面的补丁，重新基于linux-next-20230621（一切都正常工作）。
> 
> 编辑：kernel. Config：[msm8953_config.txt](https://github.com/msm8953-mainline/linux/files/11815244/msm8953_config.txt)

你好！这个分支modem可以正常工作了吗？

spasswolf, 2023-06-22:
I have not yet tried mobile data but SMS is working and calls would probably work but the fairphone-fp3 has no working audio, yet.

On the bright side, this little patch (not yet pushed) give torch led support for the fairphone-fp3 using the qcom,spmi-flash-led driver:
```
diff --git a/arch/arm64/boot/dts/qcom/pmi632.dtsi b/arch/arm64/boot/dts/qcom/pmi632.dtsi
index 4eb79e0ce40a..da3c08190241 100644
--- a/arch/arm64/boot/dts/qcom/pmi632.dtsi
+++ b/arch/arm64/boot/dts/qcom/pmi632.dtsi
@@ -151,6 +151,12 @@ pmic@3 {
 		reg = <0x3 SPMI_USID>;
 		#address-cells = <1>;
 		#size-cells = <0>;
+ 
+		pmi632_flash: led-controller@d300 {
+			compatible = "qcom,spmi-flash-led";
+			reg = <0xd300>;
+			status = "disabled";
+		};
 
 		pmi632_lpg: pwm {
 			compatible = "qcom,pmi632-lpg";
diff --git a/arch/arm64/boot/dts/qcom/sdm632-fairphone-fp3.dts b/arch/arm64/boot/dts/qcom/sdm632-fairphone-fp3.dts
index c7f6f37c1779..c4fab278ee33 100644
--- a/arch/arm64/boot/dts/qcom/sdm632-fairphone-fp3.dts
+++ b/arch/arm64/boot/dts/qcom/sdm632-fairphone-fp3.dts
@@ -142,6 +142,28 @@ &pm8953_resin {
 	linux,code = <KEY_VOLUMEDOWN>;
 };
 
+&pmi632_flash {
+	status = "okay";
+
+	flash-0 {
+		function = LED_FUNCTION_FLASH;
+		color = <LED_COLOR_ID_WHITE>;
+		led-sources = <1>;
+		led-max-microamp = <300000>;
+		flash-max-microamp = <1000000>;
+		flash-max-timeout-us = <1280000>;
+	};
+
+	flash-1 {
+		function = LED_FUNCTION_FLASH;
+		color = <LED_COLOR_ID_YELLOW>;
+		led-sources = <2>;
+		led-max-microamp = <300000>;
+		flash-max-microamp = <1000000>;
+		flash-max-timeout-us = <1280000>;
+	};
+};
+
 &pmi632_lpg {
 	status = "okay";
 
diff --git a/drivers/leds/flash/leds-qcom-flash.c b/drivers/leds/flash/leds-qcom-flash.c
index b089ca1a1901..aefaea01abbb 100644
--- a/drivers/leds/flash/leds-qcom-flash.c
+++ b/drivers/leds/flash/leds-qcom-flash.c
@@ -18,6 +18,7 @@
 #define FLASH_TYPE_VAL			0x18
 
 #define FLASH_SUBTYPE_REG		0x05
+#define FLASH_SUBTYPE_2CH_PMI632_VAL	0x05
 #define FLASH_SUBTYPE_3CH_PM8150_VAL	0x04
 #define FLASH_SUBTYPE_3CH_PMI8998_VAL	0x03
 #define FLASH_SUBTYPE_4CH_VAL		0x07
@@ -691,7 +692,9 @@ static int qcom_flash_led_probe(struct platform_device *pdev)
 		return rc;
 	}
 
-	if (val == FLASH_SUBTYPE_3CH_PM8150_VAL || val == FLASH_SUBTYPE_3CH_PMI8998_VAL) {
+	if (val == FLASH_SUBTYPE_2CH_PMI632_VAL ||
+	    val == FLASH_SUBTYPE_3CH_PM8150_VAL ||
+	    val == FLASH_SUBTYPE_3CH_PMI8998_VAL) {
 		flash_data->hw_type = QCOM_MVFLASH_3CH;
 		flash_data->max_channels = 3;
 		regs = mvflash_3ch_regs;
```

z3ntu, 2023-06-25:
@spasswolf Interesting that flash works for you, would've expected more code to be needed that's present downstream. From my notes:

> Big picture changes: driver sets POWER_SUPPLY_PROP_FLASH_ACTIVE on power supply, uses power supply for turning on power, then waits until power is on and then does the normal stuff. max-channels=2
>
> "PMI632 flash is driven by the charger in reverse boost (bharger) mode instead of the BoB regulator." (commit 1c1e21ca0c149a9eda3004c84378f92bee4a7ed7)
>
> "flash LED subtype 0x5"

Maybe it just works because we don't have smb5 charger support yet and bootloader has this power rail enabled?

spasswolf, 2023-06-28:
I just noticed that suspend (on fairphone-fp3) is working with this kernel, supend can be initiated either by pressing the power button or by selecting suspend in the gnome menu. Wakeup is a little tricky though, somtimes pressing the power button is enough but that only seems to work while the phone is connected via usb. Also sometime resuming with the power button does not work, but disconnecting and reconnecting the usb cable wakes up the phone.
 Even in the case of a succesful suspend the modem cannot be found but this is easily fixed by
`service ModemManager restart`

spasswolf, 2023-06-30:
Cherry-picking this commit has improved the resume speed, but the `gcc_usb30_master_clk status stuck at 'on'` is still there.
```
commit 250b7e1aeb02e14558e258155d67920766a3d67a
Author: Vldly <92926213+vldly@users.noreply.github.com>
Date:   Thu Jan 13 23:43:12 2022 +0000

    soc: qcom: rpmpd: add hack to speed up suspend/resume

commit 0df07fc605ffc3ab30b597de992711f5e9a
```

spasswolf, 2023-06-30:
Added a simple workaround to reduce log spamming:
```
diff --git a/drivers/input/touchscreen/himax_hx83112b.c b/drivers/input/touchscreen/himax_hx83112b.c
index 4f6609dcdef3..d1bc5543c4ee 100644
--- a/drivers/input/touchscreen/himax_hx83112b.c
+++ b/drivers/input/touchscreen/himax_hx83112b.c
@@ -238,7 +238,7 @@ static int himax_handle_input(struct himax_ts_data *ts)
        error = regmap_raw_read(ts->regmap, HIMAX_REG_READ_EVENT, &event,
                                sizeof(event));
        if (error) {
-               dev_err(&ts->client->dev, "Failed to read input event: %d\n",
+               dev_err_ratelimited(&ts->client->dev, "Failed to read input event: %d\n",
                        error);
                return error;
        }

```

z3ntu, 2023-07-01:
@spasswolf I think this touchscreen issue will ultimately be resolved with https://lore.kernel.org/linux-arm-msm/20230607215224.2067679-1-dianders@chromium.org/ (+ some extra patches for this touch driver)

spasswolf, 2023-07-02:
I've tried to integrate display following into the himax_hx83112b touchscreen driver, the first version is in branch msm8953_iommu_rebase_v2_wlan_modem_ipa_cpufreq.

This is the dmesg across a suspend/resume
```
[   32.848490] panel-djn-hx83112b 1a94000.dsi.0: Failed to set display off: -22
[   32.848545] panel-djn-hx83112b 1a94000.dsi.0: Failed to un-initialize panel: -22
[   32.901515] wlan0: deauthenticating from 54:67:51:3d:a2:e0 by local choice (Reason: 3=DEAUTH_LEAVING)
[   32.906566] msm_mdp 1a01000.display-controller: pp done time out, lm=0
[   32.973147] wcn36xx: ERROR SMD_EVENT (312) not supported
[   33.543066] PM: suspend entry (s2idle)
[   33.800742] Filesystems sync: 0.257 seconds
[   33.802280] Freezing user space processes
[   33.807677] Freezing user space processes completed (elapsed 0.003 seconds)
[   33.809324] OOM killer disabled.
[   33.816784] Freezing remaining freezable tasks
[   33.822669] Freezing remaining freezable tasks completed (elapsed 0.001 seconds)
[   33.825902] printk: Suspending console(s) (use no_console_suspend to debug)
[   33.885468] msm_mdp 1a01000.display-controller: pp done time out, lm=0
[   33.936483] msm_mdp 1a01000.display-controller: pp done time out, lm=0
[   34.120820] ------------[ cut here ]------------
[   34.120833] gcc_usb30_master_clk status stuck at 'on'
[   34.120866] WARNING: CPU: 0 PID: 1338 at drivers/clk/qcom/clk-branch.c:86 clk_branch_wait+0x140/0x158
[   34.120887] Modules linked in: q6asm_dai q6routing q6afe_dai snd_soc_core snd_compress snd_pcm snd_timer snd soundcore q6afe_clocks q6adm q6asm q6afe snd_q6dsp_common q6core leds_qcom_flash v4l2_flash_led_class v4l2_async videodev qcom_q6v5_pas qcom_wcnss_pil qcom_q6v5_mss mc qcom_pil_info qcom_q6v5 qcom_sysmon ipa qcom_common
[   34.120975] CPU: 0 PID: 1338 Comm: systemd-sleep Not tainted 6.4.0-next-20230630-debian-github-msm8953+ #802
[   34.120982] Hardware name: Fairphone 3 (DT)
[   34.120987] pstate: 600000c5 (nZCv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   34.120994] pc : clk_branch_wait+0x140/0x158
[   34.121000] lr : clk_branch_wait+0x140/0x158
[   34.121005] sp : ffff80008532ba10
[   34.121008] x29: ffff80008532ba10 x28: ffff8000822cf000 x27: ffff8000805c4518
[   34.121020] x26: 0000000000000002 x25: ffff80008215f1dc x24: ffff80008222c398
[   34.121031] x23: ffff800081dbab38 x22: 0000000000000000 x21: ffff80008045fe40
[   34.121043] x20: 0000000000000000 x19: ffff8000821f2a98 x18: fffffffffffe91c8
[   34.121054] x17: 000000040044ffff x16: 00500072b5503510 x15: 0000000000000038
[   34.121065] x14: 0000000000000000 x13: 6f27207461206b63 x12: 7574732073757461
[   34.121076] x11: fffffffffffe0000 x10: 0000000000000027 x9 : ffff80008532ba10
[   34.121087] x8 : 0000000000000027 x7 : 00000000ffffffff x6 : 0000000000000000
[   34.121098] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000027
[   34.121109] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff0000327acf80
[   34.121120] Call trace:
[   34.121123]  clk_branch_wait+0x140/0x158
[   34.121129]  clk_branch2_disable+0x2c/0x3c
[   34.121135]  clk_core_disable+0x60/0xb8
[   34.121144]  clk_disable+0x30/0x4c
[   34.121151]  dwc3_qcom_suspend.part.0+0x6c/0x238
[   34.121160]  dwc3_qcom_pm_suspend+0x3c/0x58
[   34.121166]  __device_suspend+0x108/0x34c
[   34.121175]  dpm_suspend+0x184/0x210
[   34.121182]  dpm_suspend_start+0x80/0x9c
[   34.121189]  suspend_devices_and_enter+0x6c/0x584
[   34.121200]  pm_suspend+0x1b0/0x264
[   34.121209]  state_store+0x80/0xec
[   34.121217]  kobj_attr_store+0x18/0x2c
[   34.121227]  sysfs_kf_write+0x44/0x54
[   34.121236]  kernfs_fop_write_iter+0x120/0x1ec
[   34.121244]  vfs_write+0x188/0x380
[   34.121253]  ksys_write+0x6c/0x100
[   34.121261]  __arm64_sys_write+0x1c/0x28
[   34.121269]  invoke_syscall+0x48/0x114
[   34.121278]  el0_svc_common.constprop.0+0x44/0xe4
[   34.121286]  do_el0_svc+0x38/0x98
[   34.121294]  el0_svc+0x2c/0x84
[   34.121300]  el0t_64_sync_handler+0x100/0x12c
[   34.121307]  el0t_64_sync+0x190/0x194
[   34.121315] ---[ end trace 0000000000000000 ]---
[   34.412476] msm_mdp 1a01000.display-controller: pp done time out, lm=0
[   34.463476] msm_mdp 1a01000.display-controller: pp done time out, lm=0
[   34.767783] OOM killer enabled.
[   34.774938] Restarting tasks ... done.
[   34.778936] random: crng reseeded on system resumption
[   34.783257] PM: suspend exit
[   34.914253] msm_mdp 1a01000.display-controller: pp done time out, lm=0
[   35.172429] msm_mdp 1a01000.display-controller: pp done time out, lm=0
[   36.812294] wlan0: authenticate with 54:67:51:3d:a2:e0
[   36.863101] wlan0: send auth to 54:67:51:3d:a2:e0 (try 1/3)
[   36.864727] wlan0: authenticated
[   36.869058] wlan0: associate with 54:67:51:3d:a2:e0 (try 1/3)
[   36.890092] wlan0: RX AssocResp from 54:67:51:3d:a2:e0 (capab=0x1411 status=0 aid=3)
[   36.932659] wlan0: associated
[   36.991390] wlan0: Limiting TX power to 20 (20 - 0) dBm as advertised by 54:67:51:3d:a2:e0

```

spasswolf, 2023-07-02:
This error
```
[   32.848490] panel-djn-hx83112b 1a94000.dsi.0: Failed to set display off: -22
[   32.848545] panel-djn-hx83112b 1a94000.dsi.0: Failed to un-initialize panel: -22
```
is caused by dsi_host_transfer, because msm_host->power_on is 0:
```
static ssize_t dsi_host_transfer(struct mipi_dsi_host *host,
					const struct mipi_dsi_msg *msg)
{
	struct msm_dsi_host *msm_host = to_msm_dsi_host(host);
	int ret;

	if (!msg || !msm_host->power_on) {
		printk(KERN_INFO "%s %d: msg = %px, msm_most->power_on = %d\n", __func__, __LINE__, msg, msm_host->power_on);
		return -EINVAL;
	}

	mutex_lock(&msm_host->cmd_mutex);
	ret = msm_dsi_manager_cmd_xfer(msm_host->id, msg);
	mutex_unlock(&msm_host->cmd_mutex);

	return ret;
}
```
gives
```
[   55.160548] dsi_host_transfer 1623: msg = ffff800085563848, msm_most->power_on = 0
[   55.160610] panel-djn-hx83112b 1a94000.dsi.0: Failed to set display off: -22
[   55.167769] panel-djn-hx83112b 1a94000.dsi.0: Failed to un-initialize panel: -22

```

spasswolf, 2023-07-05:
This works around both problems (the display init and the display suspend):
```
diff --git a/drivers/gpu/drm/msm/dsi/dsi_manager.c b/drivers/gpu/drm/msm/dsi/dsi_manager.c
index 28b8012a21f2..990f7c68a27c 100644
--- a/drivers/gpu/drm/msm/dsi/dsi_manager.c
+++ b/drivers/gpu/drm/msm/dsi/dsi_manager.c
@@ -550,6 +555,8 @@ int msm_dsi_manager_ext_bridge_init(u8 id)
 		if (ret < 0)
 			return ret;
 	}
+	/* swap bridges in list */
+	list_swap(&encoder->bridge_chain, encoder->bridge_chain.next);
 
 	/* The pipeline is ready, ping encoders if necessary */
 	msm_dsi_manager_set_split_display(id);
```
Reported this upstream an hope they have a better idea.

spasswolf, 2023-07-06:
Upstream had a better idea:
```
diff --git a/drivers/gpu/drm/msm/dsi/dsi_manager.c b/drivers/gpu/drm/msm/dsi/dsi_manager.c
index defebb69bbf7..28b8012a21f2 100644
--- a/drivers/gpu/drm/msm/dsi/dsi_manager.c
+++ b/drivers/gpu/drm/msm/dsi/dsi_manager.c
@@ -550,9 +550,6 @@ int msm_dsi_manager_ext_bridge_init(u8 id)
                if (ret < 0)
                        return ret;
        }
-       /* swap bridges in list, so that dsi and panel are switched
-        * on and of in the correct order on the fairphone-fp3 */
-       list_swap(&encoder->bridge_chain, encoder->bridge_chain.next);
 
        /* The pipeline is ready, ping encoders if necessary */
        msm_dsi_manager_set_split_display(id);
diff --git a/drivers/gpu/drm/panel/msm8953-generated/panel-fairphone-fp3-hx83112b.c b/drivers/gpu/drm/panel/msm8953-generated/panel-fairphone-fp3-hx83112b.c
index 8162c1e00516..851e7498fd17 100644
--- a/drivers/gpu/drm/panel/msm8953-generated/panel-fairphone-fp3-hx83112b.c
+++ b/drivers/gpu/drm/panel/msm8953-generated/panel-fairphone-fp3-hx83112b.c
@@ -393,6 +393,8 @@ static int djn_hx83112b_probe(struct mipi_dsi_device *dsi)
        drm_panel_init(&ctx->panel, dev, &djn_hx83112b_panel_funcs,
                       DRM_MODE_CONNECTOR_DSI);
 
+       ctx->panel.prepare_prev_first = true;
+
        ctx->panel.backlight = djn_hx83112b_create_backlight(dsi);
        if (IS_ERR(ctx->panel.backlight))
                return dev_err_probe(dev, PTR_ERR(ctx->panel.backlight),

```

Code is msm8953_v6.4_gpu_wlan_modem_ipa_cpufreq branch, rebased to next-20230706.

vldly, 2023-07-07:
> Upstream had a better idea:
> 
> ```
> diff --git a/drivers/gpu/drm/msm/dsi/dsi_manager.c b/drivers/gpu/drm/msm/dsi/dsi_manager.c
> index defebb69bbf7..28b8012a21f2 100644
> --- a/drivers/gpu/drm/msm/dsi/dsi_manager.c
> +++ b/drivers/gpu/drm/msm/dsi/dsi_manager.c
> @@ -550,9 +550,6 @@ int msm_dsi_manager_ext_bridge_init(u8 id)
>                 if (ret < 0)
>                         return ret;
>         }
> -       /* swap bridges in list, so that dsi and panel are switched
> -        * on and of in the correct order on the fairphone-fp3 */
> -       list_swap(&encoder->bridge_chain, encoder->bridge_chain.next);
>  
>         /* The pipeline is ready, ping encoders if necessary */
>         msm_dsi_manager_set_split_display(id);
> diff --git a/drivers/gpu/drm/panel/msm8953-generated/panel-fairphone-fp3-hx83112b.c b/drivers/gpu/drm/panel/msm8953-generated/panel-fairphone-fp3-hx83112b.c
> index 8162c1e00516..851e7498fd17 100644
> --- a/drivers/gpu/drm/panel/msm8953-generated/panel-fairphone-fp3-hx83112b.c
> +++ b/drivers/gpu/drm/panel/msm8953-generated/panel-fairphone-fp3-hx83112b.c
> @@ -393,6 +393,8 @@ static int djn_hx83112b_probe(struct mipi_dsi_device *dsi)
>         drm_panel_init(&ctx->panel, dev, &djn_hx83112b_panel_funcs,
>                        DRM_MODE_CONNECTOR_DSI);
>  
> +       ctx->panel.prepare_prev_first = true;
> +
>         ctx->panel.backlight = djn_hx83112b_create_backlight(dsi);
>         if (IS_ERR(ctx->panel.backlight))
>                 return dev_err_probe(dev, PTR_ERR(ctx->panel.backlight),
> ```
> 
> Code is in msm8953_v6.4_gpu_wlan_modem_ipa_cpufreq branch, rebased to next-20230706. Should this be added to the other panels in msm8953-generated, too?

Yes. Every panel needs it.

spasswolf, 2023-07-07:
Done and force-pushed to msm8953_v6.4_gpu_wlan_modem_ipa_cpufreq branch.

barni2000, 2024-05-11:
Hi @spasswolf  couple of stuff was upstreamed: GPU, iommu stuffs.
I think it can be closed because these are outdated patches now.

closed 2024-05-11 by barni2000