msm8953-mainline/linux #109: PMI632 vibrator

z3ntu, 2023-06-25:
A port of the `leds-qpnp-vibrator-ldo.c` driver from downstream exists at https://github.com/mtvec/msm8953-mainline-linux/tree/fp3-vibrator (HEAD: 86d0a91a66e28f22a110036e3a02a3ae74219491)

@mtvec Since the driver looks pretty clean, do you have a plan of upstreaming this?

mtvec, 2023-06-26:
Hi @z3ntu, I would love to upstream this at some point but I currently don't have a device available to test this on. So I don't know when I will have time for this unfortunately.

Feel free to take the code and upstream it yourself if you want it to be done quickly!

z3ntu, 2023-07-31:
I see there's a patch already on the mailing list from Qualcomm: https://lore.kernel.org/linux-arm-msm/20230731053712.2220898-1-quic_fenglinw@quicinc.com/T/

Not 100% sure though if it covers the same HW block, would need to check as I see no trace of "LDO" in this patch series.

PR #127 (2023-08-26): Add pmi632 vibrator driver
Since the patches have been posted on linux-arm-msm and they work fine on Fairphone 3, add the patches to our repo. Should be merged upstream sooner or later.

Also needs `CONFIG_INPUT_PM8XXX_VIBRATOR` config enabled (I use my own defconfig so not sure where to add this)

(note, not tested on 6.4.7 branch really, just directly on v6.5-rc6)

Closes #109
closed 2023-08-26 by z3ntu