msm8953-mainline/linux #117: Initial support for qpnp-qg fuel gauge on Fairphone3

spasswolf, 2023-07-16:
I've pushed an early version of a driver for the fuel gauge on the fairphone3 (earlier hardware version) to the branch msm8953_v6.4_gpu_wlan_modem_ipa_cpufreq_qg of https://github.com/spasswolf/msm8953-linux.git.
```
commit a46ef5194c8f424bc6a337d203fc10978021896c (HEAD -> msm8953_v6.4_gpu_wlan_modem_ipa_cpufreq_qg, msm8953-linux/msm8953_v6.4_gpu_wlan_modem_ipa_cpufreq_qg)
Author: Bert Karwatzki <spasswolf@web.de>
Date:   Sun Jul 16 14:38:55 2023 +0200

    power: suppyly: qcom_qg: Initial qpnp-qg fuel gauge support.
    
    This commit adds initial support for the qpnp-qg fuel gauge based
    partly on the downstream driver[1]. So far voltage and current measurement
    work and the capacity is estimated using lookup tables adapted from
    the downstream kernel[1]. Interrupt support has not yet been added.
    
    [1]: https://code.fairphone.com/projects/fairphone-3/gpl.html
    
    Signed-off-by: Bert Karwatzki <spasswolf@web.de>

commit 8396191d3bf2267549e5fb3965366e1ef5e31dae
Author: Bert Karwatzki <spasswolf@web.de>
Date:   Sun Jul 16 15:18:06 2023 +0200

    drivers: iio: adc: qcom-spmi-adc5: Add bat_therm channel.
    
    The qpnp-qg fuel gauge driver in the downstream kernel [1] for the
    fairphone-fp3 uses this channel to read the battery temperature.
    
    [1] https://code.fairphone.com/projects/fairphone-3/gpl.html
    
    Signed-off-by: Bert Karwatzki <spasswolf@web.de>

```
What works is current and voltage measuring and capacity estimation:
```
 cat /sys/class/power_supply/qcom-battery/uevent 
POWER_SUPPLY_NAME=qcom-battery
POWER_SUPPLY_TYPE=Battery
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CAPACITY=94
POWER_SUPPLY_CURRENT_NOW=-344848
POWER_SUPPLY_VOLTAGE_NOW=4300504
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=3400000
POWER_SUPPLY_VOLTAGE_MAX_DESIGN=4390000
POWER_SUPPLY_CHARGE_FULL_DESIGN=3056000
POWER_SUPPLY_TEMP=33186
```
Voltage is in µV and current in µA, negative current seems to show that we're charging (which seems to work even without charger support in kernel).

z3ntu, 2023-07-17:
Wow, that's amazing!

One sidenote I wanna note that I've seen, for the bat_therm channel is that normally downstream qpnp-vadc driver has the hardcoded NTC (Negative Temperature Coefficient) lookup table replaced by a list for the resistor that is actually found in the battery. No clue how that would be solvable in mainline since they also just have the LUT (lookup table) hardcoded in the driver.

In the best case I think you'd be checking the id resistor first to identify one of multiple possible batteries (which are defined in dt), then for this battery you take the battery data and whatnot including the thermistor data which would then get used.

Also one more question for now, did you see any code downstream specific to pm7250b (used for FG in FP4) since that also uses the same driver (in theory at least)?

spasswolf, 2023-07-17:
Just looked qpnp-qg.c in the fairhpone-fp4 kernel. Things I noticed:
(1) Instead of qpnp_vadc_read the iio_channel_read_processed functions are used.
(2) There is a lot of conditionally compiled code for PM7250
```
$ grep '[^!]defined.*CONFIG_TCT_PM7250_COMMON' -r fp4-kernel/kernel/msm-4.19/drivers/power/supply/qcom/ | wc -l
226
```

spasswolf, 2023-07-20:
To get a better estimate of the internal resistance I tried to use the method from qpnp-qg.c. Unfortunately the method seems to be inaccurate and slow here's the code which I do not really want to push yet.
```
diff --git a/drivers/power/supply/power_supply_sysfs.c b/drivers/power/supply/power_supply_sysfs.c
index 06e5b6b0e255..ad1d8b53d9e7 100644
--- a/drivers/power/supply/power_supply_sysfs.c
+++ b/drivers/power/supply/power_supply_sysfs.c
@@ -216,6 +216,7 @@ static struct power_supply_attr power_supply_attrs[] = {
 	POWER_SUPPLY_ATTR(MANUFACTURE_YEAR),
 	POWER_SUPPLY_ATTR(MANUFACTURE_MONTH),
 	POWER_SUPPLY_ATTR(MANUFACTURE_DAY),
+	POWER_SUPPLY_ATTR(ESR),
 	/* Properties of type `const char *' */
 	POWER_SUPPLY_ATTR(MODEL_NAME),
 	POWER_SUPPLY_ATTR(MANUFACTURER),
diff --git a/drivers/power/supply/qcom_qg.c b/drivers/power/supply/qcom_qg.c
index 15b530331a4c..e1e783943d3b 100644
--- a/drivers/power/supply/qcom_qg.c
+++ b/drivers/power/supply/qcom_qg.c
@@ -176,6 +176,7 @@ struct qcom_qg_chip;
 struct qcom_qg_ops {
 	int (*get_capacity)(struct qcom_qg_chip *chip, int *);
 	int (*get_temperature)(struct qcom_qg_chip *chip, int *);
+	int (*get_esr)(struct qcom_qg_chip *chip, int *);
 	int (*get_current)(struct qcom_qg_chip *chip, int *);
 	int (*get_voltage)(struct qcom_qg_chip *chip, int *);
 };
@@ -202,10 +203,15 @@ struct qcom_qg_chip {
 	/* No charger support, yet. */
 	struct power_supply *chg_psy;
 	struct mutex bus_lock;
+	// equivalent series resistance of batteryin mOhm, taken from
+	// device tree on probe but measured as in qpnp-qg.c for capacity
+	// calculation
+	int esr;
 	bool battery_missing;
 	int status;
 };
 
+
 /* from qg-util.c */
 static inline bool is_sticky_register(u32 addr)
 {
@@ -240,6 +246,23 @@ static int qcom_qg_read(struct qcom_qg_chip *chip, u32 addr, u8 *val, int len)
 	return 0;
 }
 
+/* from qg-util.c */
+int qcom_qg_read_raw_data(struct qcom_qg_chip *chip, int addr, u32 *data)
+{
+	int ret;
+	u8 reg[2] = {0};
+
+	ret = qcom_qg_read(chip, chip->qg_base + addr, &reg[0], 2);
+	if (ret < 0) {
+		dev_err(chip->dev, "Failed to read QG addr %d rc=%d\n", addr, ret);
+		return ret;
+	}
+
+	*data = reg[0] | (reg[1] << 8);
+
+	return ret;
+}
+
 /* from qg-util.c */
 static int qcom_qg_write(struct qcom_qg_chip *chip, u32 addr, u8 *val, int len)
 {
@@ -276,6 +299,27 @@ static int qcom_qg_masked_write(struct qcom_qg_chip *chip, int addr, u32 mask, u
 	return rc;
 }
 
[... 207 lines cut ...]
POWER_SUPPLY_NAME=qcom-battery
POWER_SUPPLY_TYPE=Battery
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CAPACITY=93
POWER_SUPPLY_CURRENT_NOW=-293884
POWER_SUPPLY_VOLTAGE_NOW=4323277
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=3400000
POWER_SUPPLY_VOLTAGE_MAX_DESIGN=4390000
POWER_SUPPLY_CHARGE_FULL_DESIGN=3056000
POWER_SUPPLY_TEMP=31127
POWER_SUPPLY_ESR=99
```
but the individual sample vary:
```
[ 9625.368326] qcom-qg 200f000.spmi:pmic@2:qg@4800: esr_sample[0] = 0
[ 9625.684309] qcom-qg 200f000.spmi:pmic@2:qg@4800: esr_sample[1] = 122
[ 9625.996298] qcom-qg 200f000.spmi:pmic@2:qg@4800: esr_sample[2] = 101
[ 9626.308290] qcom-qg 200f000.spmi:pmic@2:qg@4800: esr_sample[3] = 113
[ 9626.620282] qcom-qg 200f000.spmi:pmic@2:qg@4800: esr_sample[4] = 113
[ 9626.932262] qcom-qg 200f000.spmi:pmic@2:qg@4800: esr_sample[5] = 101
[ 9627.244261] qcom-qg 200f000.spmi:pmic@2:qg@4800: esr_sample[6] = 110
[ 9627.556247] qcom-qg 200f000.spmi:pmic@2:qg@4800: esr_sample[7] = 95
[ 9627.868233] qcom-qg 200f000.spmi:pmic@2:qg@4800: esr_sample[8] = 107
[ 9628.180226] qcom-qg 200f000.spmi:pmic@2:qg@4800: esr_sample[9] = 110
```

z3ntu, 2023-09-23:
Any news on this driver?

valpackett, 2026-02-07:
Both QGauge and SMB5 are available in e.g. [sm7150-mainline](https://github.com/sm7150-mainline/linux/commits/v6.19/) right now (the gauge is still not *great* lol but it's something).