msm8953-mainline/linux #142: xiaomi-vince Synaptics touchscreen not working

Ilya114, 2023-10-16:
Hello, I installed pmos using pmbootstrap on Redmi 5 Plus (Synaptics touchscreen which should work) and have no working touch. Tried plasma and phosh, another got me read-only error. Can't see dmesg because have no ip to ssh
[image]

barni2000, 2023-11-30:
Hi @Ilya114,

Maybe you have Novatek touchscreen, but it could be driver issue try to reload module first, if it is not working. Check this https://github.com/msm8953-mainline/linux/pull/135

> Sometime Novatek needs reload the module

Ilya114, 2023-12-09:
> Hi @Ilya114,
> 
> Maybe you have Novatek touchscreen, but it could be driver issue try to reload module first, if it is not working. Check this #135 You could try similar steps https://gitlab.com/postmarketOS/pmaports/-/issues/2357#note_1666785366
> 
> > Sometime Novatek needs reload the module

Thanks I will try.

barni2000, 2024-01-27:
@Ilya114 Which display do you have?

M0Rf30, 2024-01-27:
@Ilya114 Hi can I ask you what panel do you have between these values?
You should be able to see fetch the exact variant with a `cat /proc/cmdline`
or with fastboot 
```
		qcom,mdss_dsi_td4310_fhdplus_video_e7 {
			compatible = "xiaomi,td4310-fhdplus-e7";
		};
		qcom,mdss_dsi_td4310_fhdplus_video_e7_g55 {
			compatible = "xiaomi,td4310-fhdplus-e7-g55";
		};
		qcom,mdss_dsi_td4310_ebbg_fhdplus_video_e7 {
			compatible = "xiaomi,td4310-ebbg-fhdplus-e7";
		};
		qcom,mdss_dsi_nt36672_tianma_fhdplus_video_e7 {
			compatible = "xiaomi,nt36672-tianma-fhdplus-e7";
		};
		qcom,mdss_dsi_nt36672_csot_fhdplus_video_e7 {
			compatible = "xiaomi,nt36672-csot-fhdplus-e7";
			
```
The aim is to see if there are some constant bindings between a panel and a touchscreen driver
Mine is a td4310-fhdplus-e7 with synaptics touch rmi4
@barni2000 has a  td4310-fhdplus-e7-g55 with novatek nt36xx

barni2000, 2024-05-11:
Novatek support is merged, bot touchscreen is working now.

closed 2024-05-11 by barni2000

Ilya114, 2024-05-29:
Sorry for ignore, I dont own this device now XD