msm8953-mainline/linux #145: xiaomi ysl wifi connect 4WAY_HANDSHAKE_TIMEOUT

current02, 2023-11-06:
[   39.455604] wcn36xx: ERROR SMD_EVENT (312) not supported
[   41.279358] q6asm-dai c200000.remoteproc:smd-edge:apr:service@7:dais: command[0x10bdb] not expecting rsp
[   71.856710] wlan0: authenticate with 9c:9d:7e:64:0c:65
[   71.905212] wlan0: send auth to 9c:9d:7e:64:0c:65 (try 1/3)
[   71.943385] wlan0: authenticated
[   71.946826] wlan0: associate with 9c:9d:7e:64:0c:65 (try 1/3)
[   71.970867] wlan0: RX AssocResp from 9c:9d:7e:64:0c:65 (capab=0xc11 status=0 aid=4)
[   72.006011] wlan0: associated
[   78.092067] wlan0: deauthenticated from 9c:9d:7e:64:0c:65 (Reason: 15=4WAY_HANDSHAKE_TIMEOUT)
[   78.119923] wcn36xx: ERROR SMD_EVENT (312) not supported
[ 1425.822255] [drm:mdp5_irq_error_handler [msm]] *ERROR* errors: 04000000
[ 1588.133062] wlan0: authenticate with 9c:9d:7e:64:0c:65
[ 1588.187599] wlan0: send auth to 9c:9d:7e:64:0c:65 (try 1/3)
[ 1588.224597] wlan0: authenticated
[ 1588.226837] wlan0: associate with 9c:9d:7e:64:0c:65 (try 1/3)
[ 1588.251002] wlan0: RX AssocResp from 9c:9d:7e:64:0c:65 (capab=0xc11 status=0 aid=4)
[ 1588.285477] wlan0: associated
[ 1594.374641] wlan0: deauthenticated from 9c:9d:7e:64:0c:65 (Reason: 15=4WAY_HANDSHAKE_TIMEOUT)
[ 1594.415926] wcn36xx: ERROR SMD_EVENT (312) not supported
[ 1650.775880] wlan0: authenticate with 9c:9d:7e:64:0c:65
[ 1650.826507] wlan0: send auth to 9c:9d:7e:64:0c:65 (try 1/3)
[ 1650.873112] wlan0: authenticated
[ 1650.878979] wlan0: associate with 9c:9d:7e:64:0c:65 (try 1/3)
[ 1650.903031] wlan0: RX AssocResp from 9c:9d:7e:64:0c:65 (capab=0xc11 status=0 aid=4)
[ 1650.938080] wlan0: associated
[ 1657.026323] wlan0: deauthenticated from 9c:9d:7e:64:0c:65 (Reason: 15=4WAY_HANDSHAKE_TIMEOUT)
[ 1657.056196] wcn36xx: ERROR SMD_EVENT (312) not supported

barni2000, 2023-12-01:
@current02 Could you try upgrade your firmware from recovery to latest global stable?

current02, 2023-12-02:
@barni2000 yes, but it's not works,  At the same time, I tried my old router that was connected no problem, which made me feel like a driver worked incompatibility,.

barni2000, 2023-12-02:
> @barni2000 yes, but it's not works, At the same time, I tried my old router that was connected no problem, which made me feel like a driver worked incompatibility,.

than i think problem is in wcn36xx i could try search for some workaround. Maybe you could try change some settings in your router until then.

current02, 2023-12-02:
Yes at first I suspected it was the router, I thought it was the encryption, but I used Android sharing and no-secret Settings, he still didn't work, just showing the connection, but ping and other tools didn't working, How can I know the details? I'm not very good at that. I noticed that the driver matches WCN3620 but actually it is WCN3615, I didn't find this information, incidentally, the original Android system is no problem, I check the official open source system they use a very old kernel, which is a little difficult for me

> [ 1910.857032] wlan0: deauthenticating from 9c:9d:7e:64:0c:65 by local choice (Reason: 3=DEAUTH_LEAVING)
> [ 1910.900750] wcn36xx: ERROR SMD_EVENT (312) not supported
> [ 1912.301049] wlan0: authenticate with 9c:9d:7e:64:0c:65
> [ 1912.351653] wlan0: send auth to 9c:9d:7e:64:0c:65 (try 1/3)
> [ 1912.387840] wlan0: authenticated
> [ 1912.391713] wlan0: associate with 9c:9d:7e:64:0c:65 (try 1/3)
> [ 1912.415758] wlan0: RX AssocResp from 9c:9d:7e:64:0c:65 (capab=0xc01 status=0 aid=2)
> [ 1912.450316] wlan0: associated
> [ 2063.957619] wlan0: deauthenticating from 9c:9d:7e:64:0c:65 by local choice (Reason: 3=DEAUTH_LEAVING)
> [ 2064.038210] wcn36xx: ERROR SMD_EVENT (312) not supported
> [ 2064.538367] wlan0: authenticate with 9c:9d:7e:64:0c:65
> [ 2064.588417] wlan0: send auth to 9c:9d:7e:64:0c:65 (try 1/3)
> [ 2064.624683] wlan0: authenticated
> [ 2064.629321] wlan0: associate with 9c:9d:7e:64:0c:65 (try 1/3)
> [ 2064.653581] wlan0: RX AssocResp from 9c:9d:7e:64:0c:65 (capab=0xc01 status=0 aid=2)
> [ 2064.688110] wlan0: associated
> [ 2077.123948] wlan0: deauthenticating from 9c:9d:7e:64:0c:65 by local choice (Reason: 3=DEAUTH_LEAVING)
> [ 2077.170128] wcn36xx: ERROR SMD_EVENT (312) not supported
> [ 2108.467347] wlan0: authenticate with 90:de:80:73:bb:46
> [ 2108.467382] wlan0: No basic rates, using min rate instead
> [ 2108.516509] wlan0: send auth to 90:de:80:73:bb:46 (try 1/3)
> [ 2108.518806] wlan0: authenticated
> [ 2108.521590] wlan0: associate with 90:de:80:73:bb:46 (try 1/3)
> [ 2108.526251] wlan0: RX AssocResp from 90:de:80:73:bb:46 (capab=0x1431 status=0 aid=1)
> [ 2108.526274] wlan0: No basic rates, using min rate instead
> [ 2108.559832] wlan0: associated
>

current02, 2023-12-02:
Ok, thanks for your advice, incidentally this system is cool, it would be exciting to be compatible with glibc, I have the latest news that will reply to you, near time busy with work may delay a while

vipaoL, 2023-12-30:
I have the same or simillar issue too on xiaomi-markw. Sometimes I can't connect to wifi until I come very close to my router. If I remember correctly, it is a timeout error too. This bug can make the system unresponsive for some time while it tries to connect (I can't open new windows, terminal commands freeze with blinking cursor) and even crash wifi and bluetooth (wifi and bt disappear from settings and then appears again after a few seconds)

barni2000, 2024-05-11:
Hi @current02,

Is this still issue with 6.8.2?

closed 2024-05-29 by barni2000