msm8953-mainline/linux #150: xiaomi-daisy slow GPU

faveoled, 2023-12-21:
Using this kernel with PostmarketOS v23.12. Graphics is very slow. The driver seems to be loaded:
```
OpenGL vendor string: freedreno
OpenGL renderer string: FD506
OpenGL core profile version string: 3.1 Mesa 23.3.1
OpenGL core profile shading language version string: 1.40
OpenGL core profile context flags: (none)
OpenGL core profile extensions:
OpenGL version string: 3.1 Mesa 23.3.1
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.1 Mesa 23.3.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
OpenGL ES profile extensions:
```
Slow interface and web graphics:
[image]
(Not really sure if it's the right place for posting this)

barni2000, 2024-01-02:
@faveoled i think your chromium using software rendering. I have tested the gpu with supertuxkart it can reach 25 fps at second lowest preset.

barni2000, 2024-01-02:
Do you have values from android with the same test?

faveoled, 2024-01-02:
@barni2000 There were multiple green lines on chrome://gpu. Phosh And Firefox lagged as well. On Android I get 26 fps

faveoled, 2024-07-07:
Works well on pmOS v24.06.

closed 2024-07-07 by faveoled