msm8953-mainline/linux #156: Slow charging speed

vivekkj123, 2024-01-18:
Hi,

I'm using a Xiaomi Redmi 6 pro device (Sakura), and getting very slow charging speed. It takes more than 3 hours for full charge. I verified that it isn't caused by bad battery or charger because on android based custom ROMs I'm getting much faster charging speeds. 

Hope it will be resolved soon

Last but not least, Thank you for porting mainline kernel to msm8953 devices <3

Kiciuk, 2024-01-21:
Charging is limited to 1A and we are not supporting SMB chargers(No QC3)
https://github.com/msm8953-mainline/linux/blob/6.6.10/main/arch/arm64/boot/dts/qcom/msm8953-xiaomi-daisy.dts#L117
You can change it to 2 as first digit and it will be limited to 2A

alikates, 2024-01-23:
I don't remember where is it exactly but you can change the charge rate through the charger sysfs

alikates, 2024-01-23:
Although the config will be lost once you unplug the cable. You could also write a udev rule to automatically change it

closed 2024-05-11 by barni2000