msm8953-mainline/linux #166: RedMi Note 4X (Mido) after changing the screen cannot be displayed

LiuZeyu1119, 2024-01-31:
Some mido that have had their screen modules replaced cannot be displayed,these panel uses an unknown chip which can be driven normally under Android, but cannot be driven under Linux.The backlight can be turned on or off normally under Linux, but the image cannot be displayed.
[image]

Kiciuk, 2024-01-31:
cat /proc/cmdline

LiuZeyu1119, 2024-01-31:
The IC detected under aida64 is nt35532, which is the BOE panel. Because I am at work, I can only provide some other information when I return home later
[image]

Kiciuk, 2024-01-31:
it should be supported.

barni2000, 2024-01-31:
I have 2 mido with replacemenet nt35533 and they are works fin for me

LiuZeyu1119, 2024-02-01:
Some of the logs(in version 5.18.3-postmarketos-qcom-msm8953)are as follows, which I think are related to the following points.
`[    0.274929] qcom,wled 200f000.spmi:pmi8950@3:wled@d800: Only one of qcom,num-strings or qcom,enabled-strings should be set
[    0.276260] qcom,wled 200f000.spmi:pmi8950@3:wled@d800: error -ENXIO: IRQ ovp not found
...
[   16.883394] Modules linked in: apr pdr_interface venus_dec venus_enc videobuf2_dma_contig qrtr_smd s5k2xx qcom_camss qcom_q6v5_pas qrtr edt_ft5x06 snd_soc_apq8016_sbc qcom_rng leds_aw2013 snd_soc_qcom_common venus_core snd_soc_msm8916_digital qcom_wcnss_pil v4l2_fwnode qcom_fg(+) v4l2_async v4l2_mem2mem videobuf2_dma_sg qcom_q6v5_mss videobuf2_v4l2 videobuf2_memops qcom_pil_info qcom_spmi_vadc videobuf2_common rtc_pm8xxx qcom_pon qcom_q6v5 qcom_spmi_temp_alarm qcom_vadc_common reboot_mode qcom_common videodev qcom_sysmon snd_soc_msm8916_analog qmi_helpers mc i2c_qcom_cci rmtfs_mem socinfo snd_soc_aw8738 msm mdt_loader drm_dp_helper drm_kms_helper fb_sys_fops cfbfillrect syscopyarea cfbimgblt sysfillrect sysimgblt cfbcopyarea gpu_sched drm drm_panel_orientation_quirks
`
`cat /proc/cmdline
earlycon console=ttyMSM0,115200 PMOS_NO_OUTPUT_REDIRECT
`

LiuZeyu1119, 2024-02-01:
I don't think this is the real nt35532, maybe it's some kind of alternative solution from a third-party accessory manufacturer

barni2000, 2024-02-01:
> Some of the logs(in version 5.18.3-postmarketos-qcom-msm8953)are as follows, which I think are related to the following points. `[ 0.274929] qcom,wled 200f000.spmi:pmi8950@3:wled@d800: Only one of [...]

can you updated your kernel we are at 6.6.12 at the moment 5.18 is too old

LiuZeyu1119, 2024-02-01:
I have tried new version and the situation is still the same

LiuZeyu1119, 2024-02-01:
Under lk2nd, it can be displayed normally, and Tux can also display it.
https://github.com/msm8953-mainline/linux/assets/30873854/d9bc7b22-bfec-47aa-86bf-9e95bb68032d

barni2000, 2024-02-01:
try upgrade everything with ```apk update && apk ugprade -a```, and recheck your installation in my devices i have the cheapest replacement displays what i could buy for them and they are works fine.

LiuZeyu1119, 2024-02-08:
Thank you all. I updated the kernel but it still doesn't work, so I bought the original model screen and the problem was solved.

closed 2024-02-08 by LiuZeyu1119