msm8953-mainline/linux #167: xiaomi-vince: I2C power management issues

spongy-deluxe, 2024-01-31:
Hi,

currently I am experiencing issues with postmarket OS on my Xiaomi Redmi 5 Plus. I am using a dual-boot configuration with lk2nd flashed to recovery and postmarket OS flashed on an external SD card.

The issues are most likely power management related. When they set in, the phone turns the screen on, shows the lockscreen and then turns off again after a while. The dmesg output (obtained via ssh) shows `[ 8324.012680] ltrf216a 0-0053: failed to resume runtime PM: -22` repeatedly (with a frequency of 3 to 5 times per second). Finally the touchscreen becomes unresponsive and cannot be used anymore - hardware buttons still work.

I assume these issues occur because some problems with the power management. At first I suspected the bit banging of the I2C bus might be responsible but idk.

I have tested the kernels 6.6.10 and 6.7.0 (head) both yield the same behavior.

(I am not sure but the issue often occurred after I unplugged the phone from my computer's USB. Yet the issue also occurs after just setting the phone to standby.)

Do you have any clues what the underlying issue might be?

barni2000, 2024-05-11:
Hi @spongy-deluxe is this still issue with 6.8.2?

spongy-deluxe, 2024-05-15:
Hi @barni2000

yes, I still have this issue with a modified linux 6.8.2.

I have tested whether async suspend is the problem (okay, probably it was more of a "quick and dirty" test).
I just removed all calls to `device_enable_async_suspend` from the linux 6.8.2 codebase. However, this did not solve the issues (and perhaps caused even more issues in the dmesg output).

Next I'll try again with postmarket's vanilla 6.8.2-r1 and extend this answer. After that I am going to build and try the 6.9.0 kernel.

barni2000, 2024-11-05:
Switch to dpu with `msm.prefer_mdp5=false`