msm8953-mainline/linux #187: Can I boot without lk2nd?

Mr-Yuling, 2024-03-17:
I want to boot without lk2nd

vipaoL, 2024-03-17:
What device and why?
I'm not sure, but the kernel probably will boot if you add some properties like board-id to your dts. And then you'll find the display panel and touch screen not working, because lk2nd helps the kernel to select a proper node for them, you'll need to hardcode your panel driver and the dts node for your touch screen. And you'll need to carry this changes though kernel updates and recompile the kernel each time.
I see no disadvantages in lk2nd

barni2000, 2024-03-17:
You can but couple of function will be broken, apcs-clocks will not work and performance will be very bad. For display panel you will need to change panel comaptible to work etc.

closed 2024-05-11 by barni2000