msm8953-mainline/linux #189: Snapdragon 625 gpu can't be used normally in other desktop environments.

wu17481748, 2024-03-28:
Some windows on the system-level desktop are not fully displayed, but there is no such problem on the phosh desktop.

dmesg ubuntu24

[  363.832569] msm_mdp 1a01000.display-controller: [drm:a5xx_irq [msm]] *ERROR* gpu fault ring 0 fence 57b4 status E70091C3 rb 0cf0/0d70 ib1 00000000D9F18000/0e0b ib2 0000000081C11000/00ec                        [  363.832847] msm_mdp 1a01000.display-controller: [drm:recover_worker [msm]] *ERROR* 5.0.6.0: hangcheck recover!                                                                                                   [  363.833043] msm_mdp 1a01000.display-controller: [drm:recover_worker [msm]] *ERROR* 5.0.6.0: offending task: QSGRenderThread (/usr/bin/plasma-systemmonitor -session 10776300000171161382200000012040011_1711613939_807987)                                                                                                 [  363.833279] revision: 506 (5.0.6.0)                                                                    [  363.833288] rb 0: fence:    22448/22452                                                                [  363.833293] rptr:     3312                                                                             [  363.833296] rb wptr:  3440                                                                             [  363.833300] CP_SCRATCH_REG0: 0
[  363.833304] CP_SCRATCH_REG1: 0                                                                         [  363.833307] CP_SCRATCH_REG2: 22447                                                                     [  363.833311] CP_SCRATCH_REG3: 0                                                                         [  363.833314] CP_SCRATCH_REG4: 0                                                                         [  363.833317] CP_SCRATCH_REG5: 0                                                                         [  363.833321] CP_SCRATCH_REG6: 0                                                                         [  363.833325] CP_SCRATCH_REG7: 0                                                                         [  363.837914] adreno_fault_handler: 785505 callbacks suppressed
[  363.837931] *** gpu fault: iova=0000000000000000 flags=0 (0,0,0,0)                                     [  363.837945] qcom_iommu_fault: 785505 callbacks suppressed
[  363.837952] qcom-iommu-ctx 1c48000.iommu-ctx: Unhandled context fault: fsr=0x40000202, iova=0x0000000000000000, fsynr=0x2, cb=0                                                                                  [  363.837979] *** gpu fault: iova=0000000000000040 flags=0 (0,0,0,0)                                     [  363.837985] qcom-iommu-ctx 1c48000.iommu-ctx: Unhandled context fault: fsr=0x40000202, iova=0x0000000000000040, fsynr=0x2, cb=0                                                                                  [  363.837999] *** gpu fault: iova=0000000000000080 flags=0 (0,0,0,0)                                     [  363.838005] qcom-iommu-ctx 1c48000.iommu-ctx: Unhandled context fault: fsr=0x40000202, iova=0x0000000000000080, fsynr=0x2, cb=0                                                                                  [  363.838020] *** gpu fault: iova=00000000000000c0 flags=0 (0,0,0,0)
[  363.838026] qcom-iommu-ctx 1c48000.iommu-ctx: Unhandled context fault: fsr=0x40000202, iova=0x00000000000000c0, fsynr=0x2, cb=0                                                                                  [  363.838041] *** gpu fault: iova=0000000000000100 flags=0 (0,0,0,0)                                     [  363.838052] qcom-iommu-ctx 1c48000.iommu-ctx: Unhandled context fault: fsr=0x40000202, iova=0x0000000000000100, fsynr=0x2, cb=0                                                                                  [  363.838067] *** gpu fault: iova=0000000000000140 flags=0 (0,0,0,0)                                     [  363.838073] qcom-iommu-ctx 1c48000.iommu-ctx: Unhandled context fault: fsr=0x40000202, iova=0x0000000000000140, fsynr=0x2, cb=0                                                                                  [  363.838087] *** gpu fault: iova=0000000000000180 flags=0 (0,0,0,0)                                     [  363.838093] qcom-iommu-ctx 1c48000.iommu-ctx: Unhandled context fault: fsr=0x40000202, iova=0x0000000000000180, fsynr=0x2, cb=0                                                                                  [  363.838134] *** gpu fault: iova=00000000000001c0 flags=0 (0,0,0,0)                                     [  363.838141] qcom-iommu-ctx 1c48000.iommu-ctx: Unhandled context fault: fsr=0x40000202, iova=0x00000000000001c0, fsynr=0x2, cb=0                                                                                  [  363.838155] *** gpu fault: iova=0000000000000200 flags=0 (0,0,0,0)                                     [  363.838161] qcom-iommu-ctx 1c48000.iommu-ctx: Unhandled context fault: fsr=0x40000202, iova=0x0000000000000200, fsynr=0x2, cb=0                                                                                  [  363.838176] *** gpu fault: iova=0000000000000240 flags=0 (0,0,0,0)                                     [  363.838182] qcom-iommu-ctx 1c48000.iommu-ctx: Unhandled context fault: fsr=0x40000202, iova=0x0000000000000240, fsynr=0x2, cb=0

Kiciuk, 2024-03-28:
Not a place for this repo, we ain't doing mesa here

barni2000, 2024-05-11:
@wu17481748 is this still issue?
Do you use wayland or X11?
plasma-mobile and phosh works fine, they are wayland based.

closed 2024-05-29 by barni2000