msm8953-mainline/linux #192: Using mainline config to compile and report errors

wu17481748, 2024-04-26:
drivers/net/wireless/ath/ath10k/core.c:681:18: error: ‘const struct ath10k_hw_params’ has no member named ‘mcast_frame_registration’
  681 |                 .mcast_frame_registration = false,
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~
drivers/net/wireless/ath/ath10k/core.c:681:45: warning: excess elements in struct initializer
  681 |                 .mcast_frame_registration = false,
      |                                             ^~~~~
drivers/net/wireless/ath/ath10k/core.c:681:45: note: (near initialization for ‘ath10k_hw_params_list[14]’)

kaechele, 2024-05-02:
Which branch are you trying to build? I'm not getting this error on `6.8.2/main`.
The offending line is part of my patch here: https://github.com/msm8953-mainline/linux/commit/0652e4850e32504dc7f2b01a574f836a542293ac
IIRC this error existed on `6.7.5/main` but that is not the currently maintained branch, even though it is still the "default" branch in GitHub.

wu17481748, 2024-05-02:
> Which branch are you trying to build? I'm not getting this error on `6.8.2/main`. The offending line is part of my patch here: [0652e48](https://github.com/msm8953-mainline/linux/commit/0652e4850e32504dc7f2b01a574f836a542293ac) IIRC this error existed on `6.7.5/main` but that is not the currently maintained branch, even though it is still the "default" branch in GitHub.

I used the default branch of 6.7.5/main. Unexpectedly, it has been updated to the 6.8.2/main branch. thank!

closed 2024-05-11 by barni2000