msm8953-mainline/linux #205: low brightness problem on xiaomi sakura :(

L1keava, 2024-06-23:
anyone can help me? My phone is always in low brightness, no matter in recovery mode or system mode. I can't change my brightness either, but when using lk2nd or booting linux with lk2nd, the brightness is normal. I'm sad because this problem has troubled me for a long time.PS: I have tried using edl flash and fastboot flash. But they all don't work.
I'm sorry for asking an irrelevant question.

barni2000, 2024-07-25:
@L1keava is your phone brightness working fine on postmarketos but wrong on android?

L1keava, 2024-07-26:
> @L1keava is your phone brightness working fine on postmarketos but wrong on android?

Yes, you are absolutely right, can you help me solve this problem plz.

barni2000, 2024-07-26:
Unfortunately i cannot help with android you can try to install a custom ROM and check what happens in twrp, maybe firmware upgrade will help idk.

closed 2024-07-26 by barni2000