msm8953-mainline/linux #220: Moto G5 Plus (XT1683): touchscreen doesn't work

NickRMD, 2024-12-13:
I have noticed that after installing postmarketOS on my G5 Plus the touchscreen won't work. The display works normally and displays the DE (Plasma-mobile, wayland) correctly, the sound works, the wifi and bluetooth and battery gauge, but the touchscreen don't work.

The possible reason seems to be a misconfigured dts file for the device:
After installing I went to the postmarketOS and postmarketOS Mainline Matrix room and started asking what could be the problem, it seems like the dts file for the G5 Plus has some misconfigured regulators as from the [original dts](https://github.com/MotorolaMobilityLLC/kernel-msm/blob/nougat-7.0.0-release-potter/arch/arm/boot/dts/qcom/msm8953-potter.dtsi).
You could see from the logs (dmesg files) attached that the touchscreen don't seen to initialize (as from the error -6 [No such device or address](https://gist.github.com/greggyNapalm/2413028)).

[TWRP dmesg full log:](https://github.com/user-attachments/files/18131585/twrp.dmesg.txt) initializes correctly;
[postmarketOS dmesg full log:](https://github.com/user-attachments/files/18131617/postmarketos.dmesg.txt) doesn't initialize.

I would be grateful if anyone could help and I'm willing to help with anything if needed since I have the phone in hand and can help with testing and such.

barni2000, 2024-12-15:
I have tried to correct it but still not works.
```
-               interrupts = <65 IRQ_TYPE_EDGE_FALLING>;
+               interrupts-extended = <&tlmm 65 IRQ_TYPE_LEVEL_LOW>;

+               reset-gpios = <&tlmm 64 GPIO_ACTIVE_LOW>;
                pinctrl-names = "default";
-               pinctrl-0 = <&ts_reset>;
+               pinctrl-0 = <&ts_int_reset>;

                vdd-supply = <&pm8953_l22>;
                vio-supply = <&pm8953_l6>;
@@ -282,7 +284,6 @@ pm8953_l5: l5 {
                pm8953_l6: l6 {
                        regulator-min-microvolt = <1800000>;
                        regulator-max-microvolt = <1800000>;
-                       regulator-always-on;
                };

                pm8953_l7: l7 {
@@ -341,8 +342,7 @@ pm8953_l19: l19 {
                };

                pm8953_l22: l22 {
-                       regulator-always-on;
-                       regulator-min-microvolt = <2800000>;
+                       regulator-min-microvolt = <2850000>;
                        regulator-max-microvolt = <2850000>;
                };

@@ -386,8 +386,8 @@ &sound_card {
 &tlmm {
        gpio-reserved-ranges = <1 2>, <96 4>, <111 1>, <126 1>;

-       ts_reset: ts-reset-state {
-               pins = "gpio64";
+       ts_int_reset: ts-int-reset-state {
+               pins = "gpio64", "gpio65";
                function = "gpio";
                drive-strength = <2>;
                bias-pull-up;
```

barni2000, 2024-12-18:
I have find out issue is with missing reset handling from the driver.
Adding a "fake" node with other touchscreen driver what handling reset makes it work,
so synaptics driver should be modified to handle `reset-gpios`
```
touchscreen@19 {
               compatible = "edt,edt-ft5406", "edt,edt-ft5x06";
               reg = <0x19>;

               reset-gpios = <&tlmm 64 GPIO_ACTIVE_LOW>;

               status = "okay";
};
```

barni2000, 2025-01-13:
Here are some better WIP soltion
https://github.com/kaechele/msm8953-mainline-linux/commit/fb11ec8ba8ff193002bbcbd72b13d0e0023c8f72
https://github.com/kaechele/msm8953-mainline-linux/commit/19b855487be354139ddc9389fa940f75a07c3e08

PR #223 (2025-01-20): Fix touchscreen initialization on Motorola G5 Plus (Potter)
Fixes #220 

Let's test this here before I send it upstream.
closed 2025-01-20 by barni2000